realsenseai/librealsense #4430: Can I use PXCFaceModule with D435 to try Face Tracking?

coldcoda, 2019-07-15:
Can I use PXCFaceModule with D435 to try Face Tracking? And get   face location detection, landmark detection? How can I get the PXCFaceModule ?

MartyG-RealSense, 2019-07-15:
The PXCFaceModule and other modules from the 2016 RealSense SDK are unfortunately not compatible with modern '400 Series' RealSense cameras such as D415 and D435.

The face features that you are interested in are available for the D415 and D435 in a commercial software package called Nuitrack SDK.  It costs $39.99 a year, and also has a free trial version.

https://nuitrack.com/

https://download.3divi.com/Nuitrack/doc/UnityFaceTracking_page.html

dorodnic, 2019-07-16:
You can get face detection and landmarks for free from any camera using opencv and dlib:
[facial-landmarks-dlib-opencv-python](vhttps://www.pyimagesearch.com/2017/04/03/facial-landmarks-dlib-opencv-python/)

closed 2019-07-22 by dorodnic