realsenseai/librealsense #4431: .deb package pipeline

dorintt, 2019-07-15:
Debian package for amd64 is available and this is great, but for arm32 there is no deb package. Considering that compilation from source is a quite slow and troublesome process for arm32 platforms, it would be really nice to have a deb package for it as well.

Or at least provide the scripts for generating .deb package ourselves, so we can compile once and install many times. Why is it a big deal to opensource these scripts?

dorodnic, 2019-07-16:
It is not a big deal, and it will be done, in a deliberate manner. 
Seems to me that someone who is serious about cross-compiling, hosting and publishing packages for Arm architecture, would not be stopped by generating .deb files from open-source CMake project.

Two related tasks that will benefit Arm users (and are being worked on) are:
a. Expanding the LibUVC backend to reduce the need for Kernel patching
b. Cross-compiling and publishing Debians from the official servers

These are, in my opinion, preferable to several community maintained 3rd-party repositories.

dorintt, 2019-07-16:
Thanks, I understand.
It is definitely possible to adapt the cmake project, I know there is cpack but I'm not too familiar with it and I thought why have duplicate work if the scripts are already written.

I don't intend to publish on debian repositories, I only want to speed up the internal process of installing on new devices. Of course intel repositories are fine as well.

dorodnic, 2019-07-16:
I can ensure you that we understand the importance of supporting wide range of platforms and architectures and working on it.

sjdrc, 2019-09-08:
Having CPack generate proper packages would be useful in the interim, as currently we need to do source builds to compile CUDA support for the jetson.

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Hi @dorintt This change request is approved. We will implement it.

icefoxen, 2020-10-29:
Hi, any news on this?  I'm interested in building .deb's for the latest Realsense master/release versions for Jetson TX2.  If it helps I can work on troubleshooting and making PR's for any .deb script build issues for this platform.

RealSenseSupport, 2020-11-23:
This is still work in progress.
However, the latest v2.40.0 release has added kernel patches for V4L driver for Jetson. 
Please check: https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

seebq, 2021-01-06:
Any news on providing the scripts for generating .deb package ourselves?  As we move to a more production type environment, "compiling from source" on field units doesn't sit well with us.

Happy to help with this as I agree that "generating .deb files from open-source CMake project" is probably the way to go.

We are very interested in this as we have a custom kernel and the librealsense-dkms scripts do not work with anything other than the stock ubuntu lts kernels.

seebq, 2021-03-01:
Today I am manually building and guessing at the debian build scripts to package up the latest release https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

Sure would be nice to get that sweet `debian/` directory y'all have somewhere and the build instructions so we can build and distribute this library against our custom kernel (that also has all the same dkms patches).  Building from source is not an option on our production robots in the field.

Any chance you're still considering open sourcing the debian build scripts?

MartyG-RealSense, 2022-04-23:
Hi everyone, I would like to provide an update about this case.

Intel opened a Change Request report in December 2019 to consider the provision of an Arm32 Debian package or alternatively a script for generating the Debian package.

This Change Request was closed in June 2021 with the decision to not proceed further with this approach.  It was decided that a different approach would be taken - the provision of Debian Arm packages through the next generation of the RealSense ethernet networking system.  That work on the next-generation networking system is ongoing at the time of writing this and I do not currently have a time estimate for when it will be released. 

This case will be closed in 7 days from the time of writing this if further comments are not received.  Thanks very much for your patience!

closed 2022-05-04 by MartyG-RealSense