realsenseai/librealsense #4432: Multiple camera example error

a17sans, 2019-07-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Win 10  | 
| Platform                                 | PC |
| Language                            |  python | 

### Issue Description

I tried to use the multi camera example : [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam)

And I got the following error : 

` TypeError : poll_for_frames() : incompatible function arguments. The following argument types are supported : 1. (self: pyrealsense2.pyrealsense2.pipeline) -> pyrealsense2.pyrealsense2.composite_frame
Invoked with : <pyrealsense2.pyrealsense2.pipeline object at 0x00000255FFCFCBC8>, <pyrealsense2.pyrealsense2.composite_frame object at 0x00000255FFDF7110>
 `

I understand that I call the function `poll_for_frames()` with bad arguments, as the code is : 
`device.pipeline.poll_for_frames(frameset)` (frameset being of type composite_frame) but the `poll_for_frames()` function does not want any argument. 

Has anyone got the same error ? 

Thank you very much !

a17sans, 2019-07-16:
Here is the code on the example wich gives the error :
`frameset = rs.composite_frame(rs.frame())`
`device.pipeline.poll_for_frames(frameset)`

But as `poll_for_frames()` returns a `composite_frame` object, I tried this : 
`frameset = rs.composite_frame(rs.frame())`
`newFrameset = device.pipeline.poll_for_frames()`

But frameset and newFrameset have a length of 0 ...

RealSenseCustomerSupport, 2019-08-01:
----------------------------------------------

Please check if the camera model and the video stream configuration are set correctly.
https://github.com/IntelRealSense/librealsense/blob/d19829788008b8e000870895a068f0c43d58895a/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L305

RealSenseCustomerSupport, 2019-08-09:
----------------------------------------------

Hi @a17sans, has this issue been resolved ? Can we close it?

closed 2019-08-13 by RealSenseCustomerSupport