realsenseai/librealsense #4434: Android support  for Intel Realsense D435i cmaera

AmanTiwari1503, 2019-07-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00| 
| Operating System & Version |   Android Pie (API 28)  | 
| Platform                                 | Android  |
| SDK Version                            |  2.23.0  | 
| Language                            |  Java (Android Studio) | 
| Segment			|  Smartphone  | 

### Issue Description
When I am making use of the Intel RealSense D435i in order to run apps such as RSCapture or RSCamera or other such apps (provided by default in the github repository in wrappers/android/examples) the app is not working correctly in the sense that the video output from the camera is not getting displayed on the screen on the phone. I checked the error in the MainActivity.java file of the default apps and found the error getting displayed as 
"streaming error = RS2_EXCEPTION_TYPE_WRONG_API_CALL_SEQUENCE"

Is the support for D435i still not provided by Intel or is there some other kind of error?
Kindly help me on this matter.

PS. All the apps are working perfectly when tested with the Intel Realsense D415 camera

MartyG-RealSense, 2019-07-15:
According to the notes on the SDK 'Releases' page, Android support for the D435i is still under development to be included in a future SDK release.

https://github.com/IntelRealSense/librealsense/releases/

AmanTiwari1503, 2019-07-15:
Thanks for giving the update @MartyG-RealSense

AmanTiwari1503, 2019-07-26:
> Thanks for giving the update @MartyG-RealSense .
> When can I expect the android support for D435i to be released?

Any comments?

MartyG-RealSense, 2019-07-26:
I'm not part of the development team, so I do not have knowledge of a time-frame for when Android support for the D435i will be ready.  One of the RealSense development team on this forum such as @dorodnic may be able to provide guidance.

kinectsources, 2019-07-26:
Does Intel realsense d series support with Google AR platform called ARCore (non ARcore supported phones) like tango like devices? Or is there any alternative libraries to develop AR apps?

MartyG-RealSense, 2019-07-26:
Nuitrack SDK (the 400 Series-compatible skeleton and face tracking commercial software) has a tutorial for the Unity game engine that uses ARCore.  Although the article does not mention RealSense, Nuitrack SDK works excellently with the 400 Series cameras in Unity (it is one of the 'supported devices' mentioned in the tutorial's requirements list).

https://download.3divi.com/Nuitrack/doc/UnityARCore_page.html

There are a range of AR SDks available (googling for 'ar sdks' gives some suggestions).  History has taught me that sticking with the larger engines such as ARCore, Unity and Unreal Engine usually works out for the best though in terms of minimizing frustration.

ev-mp, 2019-07-29:
Duplicate of #4384

dorodnic, 2019-08-08:
Please see #4384

closed 2019-08-08 by dorodnic

MartyG-RealSense, 2019-08-21:
@AmanTiwari1503     I just wanted to update your case to let you know that D435i support for Android has apparently been added in the new SDK version 2.26.0, according to the Releases web-page.