realsenseai/librealsense #4435: Announcement: introducing the RealSense SR305 camera

MartyG-RealSense, 2019-07-15:
Hi everyone, 

A new version of the RealSense SR300 called the SR305 is now listed in Intel's online Click store.  It will begin shipping in the week of July 29 2019 and contains an SR300 depth module in a new slimline casing with mounting bracket and tripod thread-hole.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-sr305.html

ShettyHarapanahalli, 2019-07-16:
Hi,

Is there any comparison between D415, D435i, and SR305 depth cameras. 

Can you share a few examples of applications where each one of these cameras will be of a better fit?
 
Is SR305 IP68 rated camera?

MartyG-RealSense, 2019-07-16:
The link below has a blog article comparing the different types of 400 Series camera.

https://www.intelrealsense.com/which-device-is-right-for-you/

In regard to the SR305, it is difficult to make a direct comparison to the 400 Series because it uses a type of camera technology called Coded Light, whereas the 400 Series uses a different technology called Stereo.  An information page for the SR305, which includes example applications, can be found here:

https://www.intelrealsense.com/coded-light/

IP68 - the rating for whether a product is waterproof - does not apply to the 400 Series cameras (they can work underwater but have to be in a protective container).  I wouldn't think that the SR305 was waterproofed either, as it is not an environment that a RealSense camera would be commonly used in.

rcohe032, 2019-07-16:
What is the difference compared to the SR300? Thanks!

MartyG-RealSense, 2019-07-16:
The SR300's technology is a couple of years older than the 400 Series, so one of the most obvious differences is in the maximum depth and IR resolution that it can display (about half that of the 400 Series at 640x480).  It also has a maximum depth sensing range of around 1.5 m, compared to the 10 m of the 400 Series (though the accuracy of the 400 Series cameras drifts noticeably after about 3 m).  

It has some advantages though, such as better accuracy than the 400 Series at very close range scanning.  It has a dual microphone array (microphones were removed from the 400 Series).  Also, the coded light tech of the SR300 supports a 'confidence threshold' feature that can decide whether the depth of a pixel is marked as valid or invalid.  Stereo cameras like the 400 Series cannot do this.

dorodnic, 2019-07-16:
The two main differences between SR300 and SR305 are mechanical housing and price. 
The SR305 housing is more in-line with our current line of product, and intended to be more suitable  for applications in robotics (with more compact rectangular design, mounting screws and tripod mount)
It was redesigned as a good entry level depth camera, and in general should be more affordable than some of the older third-party SR300 peripherals.
In terms of the software stack, this product will not be officially supported by the 2016 RealSense SDK, but will be fully supported by librealsense2 (in terms of platforms and wrappers)

zxli, 2019-07-17:
Does SR305 support Windows Hello facial recognition?

MartyG-RealSense, 2019-07-17:
Since the SR305 contains an SR300 module board, I would speculate that it should be able to in theory, since the SR300 supported it.  In practice though, the reliability of Windows Hello with an SR300 would not seem good if RealSense forum help requests were used as an indicator.   

The trouble with this approach is that people rarely post a message about something if it works fine for them, so there is no way to tell if the reports of problems reported represented the majority of RealSense Windows Hello users or a very small reported minority among a majority that it was successful for.

My advice would be to purchase the camera if you want it for a depth sensing project, and treat Windows Hello as an added bonus if it works on your particular computer.  That way, you avoid being disappointed if Windows Hello does not work.

dorodnic, 2019-07-17:
The SR305 will not support Windows Hello.

ShettyHarapanahalli, 2019-07-17:
Can we do SLAM with this camera, just like T265 ?

dorodnic, 2019-07-22:
No, it's functionally similar to other RealSense **depth** cameras.

NXTwoThou, 2019-08-01:
Just found this after reading the announcements a few weeks ago and deciding to pull the trigger yesterday.  After getting the SDK update email today I noticed that there wasn't a separate firmware update available for the SR305.  From reading the postings in here, does this mean there's no difference in the SR300 and SR305 other than the case?

MartyG-RealSense, 2019-08-01:
Thanks so much for letting me know about the SDK update!  I check the Releases page daily for progress, yet the next version always comes out exactly when I'm not looking.   :)

The SR305 contains an SR300 camera board inside it.  As far as I am aware, the casing is the main difference, and it is being pitched and priced as an entry-level member of the RealSense camera family.  

It is recommended that it is used with RealSense SDK 2.0, and that use with the old 2016 Windows SDK is "not officially supported".  I have not received confirmation from the RealSense team though whether it is incapable of being used with the 2016 SDK, or whether they are simply discouraging people from using that old SDK because it is no longer updated.

I do not have information about the SR300 / SR305 firmware situation.  I know that the recommended firmware for SR300 was changed in this new release.  One of the Intel staff on this forum such as @dorodnic is best placed to comment on this.

richardw347, 2019-08-13:
Hi! Does anyone know if/when a CAD model for the SR305 will be released? 

Thanks

MartyG-RealSense, 2019-08-13:
I have not come across a CAD model for the SR305 yet, though details of the SR305 have now been added to the SR300 data sheet document, and mechanical drawings for the SR305 can be found on pages 47 and 48.

https://www.intelrealsense.com/wp-content/uploads/2019/07/RealSense_SR30x_Product_Datasheet_Rev_002.pdf?_ga=2.174069345.211630113.1564524199-725826302.1555336840

jakemin, 2019-08-28:
There is a way to support Windows Hello and 2016 RealSense SDK on SR305.
Run "Install Recommended Firmware" in Intel RealSense Viewer V2.26.0.
This will downgrade to V3.21.0.0 and disconnect.
When you run Intel RealSense Viewer V2.26.0. Again, you will be prompted for a recommended firmware update(V3.26.1.0).
However, it is important to note that the model will change from SR305 to SR300 when the above method is executed.
The SR305's firmware is 3.27.3.0 and the SR300's firmware is 3.26.1.0.

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@richardw347 - CAD model for SR305 has been released on the following page/site:

https://dev.intelrealsense.com/docs/cad-files

Thanks

MartyG-RealSense, 2019-09-20:
@RealSenseCustomerSupport   Thanks so much for providing the SR305 CAD model link.  I will post it on the Intel Support forums.

sls3d, 2019-09-22:
I already have a SR300, using it to 3d scanning. Do i understand it correctly, that there are no real improvements between SR300 and SR305, except price and hull design?
No improvements to performance or accuracy for 3d scanning?

Thank you

MartyG-RealSense, 2019-09-22:
It is the same SR300 specification camera inside the new casing.  Aside from the new casing for easier mounting of the camera, the SR305 is aimed at pitching the SR300 technology as an affordable entry-level introduction to RealSense technology.

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Hi @MartyG-RealSense - Is there anything further on this announcement thread that is needed or can we close this one out?

MartyG-RealSense, 2019-10-04:
You can close it, yes.  Dorodnic's policy with announcements is that once they disappear off the bottom of the front page of the forum then they can be closed.  Thanks!

RealSenseCustomerSupport, 2019-10-04:
----------------------------------------------

Good policy :-) Thanks

MartyG-RealSense, 2019-10-04:
You're very welcome.  I just did a search for past announcements and closed 6 of them for you.

closed 2019-10-04 by RealSenseCustomerSupport