realsenseai/librealsense #4436: How to patch V4L, legacy SDK for R200 and UP Board

k3nden, 2019-07-15:
I purchased the Realsense R200 development kit and having problems with installing and using the camera.
Installed on the UP Board is Ubuntu16.04.6, kernel is 14.15-37, and Kinetic ROS.
I followed the instructions for the legacy builds. https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md

All was well until step 3: I do not know what to do regarding Video4Linux backend. Do I need patches? if so which ones?
cpp-config-ui - No device found.

At present I am not sure if I am chasing an install issue or a camera that does not work.
Any advice would be appreciated.
TIA

dorodnic, 2019-07-16:
Hi @k3nden 
There is no need to patch modern kernels for R200.
I suspect it is not detecting the device due to missing UDEV rules (see the remainder of the guide).
Check if you can see the device using `lsusb` and `dmesg`. Let us know if you still encounter problems.

k3nden, 2019-07-17:
Thank you for your quick reply. I have provided the output of both the dmesg and lsusb.  For some reason the camera is not initializing?  

> sudo dmesg | tail -n 50
[  187.040789] Linux video capture interface: v2.00
[  187.095399] usbcore: registered new interface driver uvcvideo
[  187.095402] USB Video Class driver (1.1.1)
[ 3792.566032] mmc0: Tuning timeout, falling back to fixed sampling clock
[10991.763521] mmc0: Tuning timeout, falling back to fixed sampling clock
[17930.029948] perf: interrupt took too long (2530 > 2500), lowering kernel.perf_event_max_sample_rate to 79000
[23559.410357] perf: interrupt took too long (3216 > 3162), lowering kernel.perf_event_max_sample_rate to 62000
[30546.441027] perf: interrupt took too long (4025 > 4020), lowering kernel.perf_event_max_sample_rate to 49500
[37341.578245] mmc0: Tuning timeout, falling back to fixed sampling clock
[37972.001724] mmc0: Tuning timeout, falling back to fixed sampling clock
[40169.065825] perf: interrupt took too long (5042 > 5031), lowering kernel.perf_event_max_sample_rate to 39500
[46994.012862] mmc0: Tuning timeout, falling back to fixed sampling clock
[52934.799329] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[52934.823168] usb 2-1: New USB device found, idVendor=8086, idProduct=0a80
[52934.823185] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52934.823195] usb 2-1: Product: Intel RealSense 3D Camera R200
[52934.823205] usb 2-1: Manufacturer: Intel Corp
[52934.823214] usb 2-1: SerialNumber: SN_2521007997
[52934.829138] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[52934.834617] uvcvideo 2-1:1.0: Entity type for entity Extension 2 was not initialized!
[52934.834624] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[52934.836179] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[52934.840869] uvcvideo: Unable to create debugfs 2-3 directory.
[52934.841257] uvcvideo 2-1:1.2: Entity type for entity Camera 4 was not initialized!
[52934.841832] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/input/input10
[52934.844717] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[52934.848190] uvcvideo: Unable to create debugfs 2-3 directory.
[52934.848715] uvcvideo 2-1:1.4: Entity type for entity Processing 7 was not initialized!
[52934.848723] uvcvideo 2-1:1.4: Entity type for entity Camera 6 was not initialized!
[52934.849279] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input11
[52935.711132] usb 2-1: USB disconnect, device number 3
[52935.991549] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[52936.014813] usb 2-1: New USB device found, idVendor=8086, idProduct=0a80
[52936.014828] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52936.014837] usb 2-1: Product: Intel RealSense 3D Camera R200
[52936.014844] usb 2-1: Manufacturer: Intel Corp
[52936.014852] usb 2-1: SerialNumber: SN_2521007997
[52936.019323] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[52936.024638] uvcvideo 2-1:1.0: Entity type for entity Extension 2 was not initialized!
[52936.024655] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[52936.026468] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[52936.030945] uvcvideo: Unable to create debugfs 2-4 directory.
[52936.031311] uvcvideo 2-1:1.2: Entity type for entity Camera 4 was not initialized!
[52936.031557] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/input/input12
[52936.032972] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[52936.036325] uvcvideo: Unable to create debugfs 2-4 directory.
[52936.036721] uvcvideo 2-1:1.4: Entity type for entity Processing 7 was not initialized!
[52936.036729] uvcvideo 2-1:1.4: Entity type for entity Camera 6 was not initialized!
[52936.037085] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input13
[52936.306956] usb 2-1: USB disconnect, device number 4

LSUSB
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0424:2530 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:4603 Standard Microsystems Corp. 
Bus 001 Device 002: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

RealSenseCustomerSupport, 2019-08-08:
----------------------------------------------

In case this issue still persists, please check the #195.
https://github.com/IntelRealSense/librealsense/issues/195

closed 2019-08-15 by RealSenseCustomerSupport