realsenseai/librealsense #4437: Where is th pxcsensemanager?

llrtt, 2019-07-16:
How can i find the pxcsensemanger.h? I've seen many instances containing this, but i never see one from this github.

MartyG-RealSense, 2019-07-16:
pxcsensemanager dates back to the old 2016 RealSense SDK for the F200, R200 and SR300 camera models.  If you are not using one of those old models or are not using the 2016 SDK then you shouldn't worry about pxcsensemanager.

llrtt, 2019-07-16:
How can use similar function as gesture recognition with current  RealSense SDK? @ @MartyG-RealSense

MartyG-RealSense, 2019-07-16:
Hand, body and face tracking is available for the D415 and D435 in a commercial software package called Nuitrack SDK. It costs $39.99 a year, and also has a free trial version.

https://nuitrack.com/

https://download.3divi.com/Nuitrack/doc/UnityFaceTracking_page.html

Dorodnic the RealSense SDK Manager offered the following additional suggestion to another user who wanted to do face tracking, 

**********

You can get face detection and landmarks for free from any camera using opencv and dlib:
facial-landmarks-dlib-opencv-python

llrtt, 2019-07-16:
Do you know where to get  2016 RealSense SDK ? I'm using a  SR300 camera and want to write an application to recognize  some  gestures.  @MartyG-RealSense

MartyG-RealSense, 2019-07-16:
Version '2016 R3' of the old SDK is available from the link below.   It is split up into a core 'Essentials' module and a number of optional additional modules such as 'Hand Tracking' and 'Face Tracking'.  You should install the Essentials module first, and then after that install the optional modules for the features that your project needs.

https://software.intel.com/en-us/realsense-sdk-windows-eol

closed 2019-07-21 by dorodnic