realsenseai/librealsense #4439: 3D corridinates from realsense Python interface, i.e.;  rs.format.XYZ32F is format support

ShettyHarapanahalli, 2019-07-16:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|745412070314|
|**Firmware Version**|05.10.03.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

I am Using real sense for a robotic application, I need to get the 3D coordinates of the image, I see the formate to get 3D data from the sensor is "rs.format.XYZ32F", but when I use this in python interface I get the below error - 

AttributeError: type object 'pyrealsense2.pyrealsense2.format' has no attribute 'XYZ32F'

Can you please help me get 3D coordinates of the image, I am aware of "rs2_deproject_pixel_to_point"  feature but I need to process 3D data from multiple points so the support format like XYZ32F or equivalent will be of good help.

ShettyHarapanahalli, 2019-07-17:
Just wondering if we can do something like "Pixy2 CMUcam5" on any Realsense device, where we deploy ML model (instead of color-based filtering algorithm as in pixy), do processing on data and generate simpler/text outputs for further processing down the line.

lramati, 2019-08-21:
As can be seen in [the documentation](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.format.html#pyrealsense2.format) the correct name for this enum value in python is `rs.format.xyz32f`.
However, if you want a pointcloud from the D435, you'll need to enable the depth (and optionally color) and then pass these to the `rs.pointcloud` processing block

closed 2019-09-03 by dorodnic