realsenseai/librealsense #4440: How does Depth Quality Tool know the angle without IMU

rcohe032, 2019-07-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.11.06.250) | 
| Operating System & Version |   {Win (10)| 
| Kernel Version (Linux Only)    |                                | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.21.0 }                          | 
| Language                            |  {C++ OpenCV }                          | 
| Segment			|                     | 

### Issue Description
<Describe your issue / question / feature request / etc..>

How does the Depth Quality Tool measure the angle of my D415 to the ground without an IMU? Is it possible to implement this with the sdk in c++? 

Thanks!

agrunnet, 2019-07-16:
It does a plane fit.

rcohe032, 2019-07-17:
Thank you! 

I am trying to do a plane fit to calculate the angle myself. I have moved to Matlab to do it, with my device at an angle of around 14 degrees from left to right (the left side is lower than the right side of the device).

I have my pointcloud set, and cropped out the data that isn't properly read (0 or noise). I am left with the first image (blue figure). 
[image]

I use a Matlab function to fit a mesh plane to the data, which ends up fitting very well. 
[image]

I created this mesh by solving for x in DM*B= Z where DM is my x and y coordinates and Z is my z coordinates. B then is the coefficients in my plane equation. 

`DM = [vertices(:,1), vertices(:,2), ones(size(vertices(:,3)))];                             % x,y,1
  B = DM\vertices(:,3);                                               % Solve for coefficients
%grid with size x and y 
[X,Y] = meshgrid(linspace(min(vertices(:,1)),max(vertices(:,1)),50),linspace(min(vertices(:,2)),max(vertices(:,2)),50));
Z = B(1)*X + B(2)*Y + B(3)*ones(size(X)); %equation of the plane`

I then got the angle between this plane and the camera's plane. I took the camera's plane normal to be parallel to the z axis (0,0,1). 

`v= B/norm(B); %normal vector to fitted plane

 v2= [0;0;1]; %camera's plane
       
 angle = acosd(dot(v,v2)/(norm(v)*norm(v2)));`

I keep getting that the angle between the 2 planes is 41 degrees, which is way off from what the Depth Tool is saying. Am I taking the right plane for the camera's plane? 
Here is a picture of the Depth Tool 
[image]

Thanks for your help!

rcohe032, 2019-07-18:
I solved it! I was taking the wrong coefficients in B. The normal should be B(1), B(2), and 1.

closed 2019-07-18 by rcohe032