realsenseai/librealsense #4443: Using librealsense with catkin

MCSuperior, 2019-07-17:
I try to use librealsense in a ROS environment on Ubuntu 16.04, but without using the Intel-ROS bridge. It is not possible to make my program with Catkin, although I followed the instructions on how to alter the CMakeList for it. The problem is that Catkin does not find anything related to "realsense2". The CMakeList which I used is shown below. Note that other libraries are also used.
Before, I have installed librealsense, first using the instructions on how to get the pre-compiled packages. After this issue appeared, I also built the package as described in the manual Linux installation guide, which did not work either.
I also stumbled upon a guide which is given on the ROS website [(Link),](http://wiki.ros.org/realsense_camera/Tutorials/Building_librealsense_from_Sources) and tried to follow the instructions for the librealsense part. The problem was that I was not able to make the librealsense package, as the error

`Package librealsense not found in the workspace`

implies.

What to do?

CMakeList:

```
cmake_minimum_required(VERSION 3.1.0)
project(xxxx)
add_compile_options(-std=c++11)
add_compile_options(-lrealsense2)
set(PCL_DIR "/usr/lib/x86_64-linux-gnu/cmake/pcl/")
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  cv_bridge
  pcl_ros
  realsense2 REQUIRED
  PCL REQUIRED
  OpenCV REQUIRED PATHS /usr/local/include/opencv2 NO_DEFAULT_PATH
)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)

include_directories(
  ${catkin_INCLUDE_DIRS}
  ${realsense_INCLUDE_DIR}
  ${PCL_INCLUDE_DIRS}
  ${OpenCV_INCLUDE_DIRS}
)

link_directories(
  ${PCL_LIBRARY_DIRS}
  ${OpenCV_LIBRARY_DIRS}
)

add_definitions(
    ${PCL_DEFINITIONS}
)

add_executable(xxx src/xxx.cpp)
target_link_libraries(
    ${catkin_LIBRARIES}
    ${realsense2_LIBRARY}
    ${OpenCV_LIBS}
    )
```

eMrazSVK, 2019-07-24:
Only thing I have in my CMakeLists.txt within my package in my catkin_ws is this:
`find_package(realsense2 2.22.0)`
This line is before
```
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  image_transport
  cv_bridge
  OpenCV
  message_generation
)
```

Try to do it like me, so you have:
```
find_package(realsense2 2.22.0)
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  image_transport
  cv_bridge
  OpenCV
  message_generation
)
```
Note: I guess you do not have to specify realsense2 version.

Also I think, that you do not want to have librealsense as a Catkin component. They neither do it in official [realsense-ros wrapper](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/CMakeLists.txt#L37) (exactly here is how they do it).

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@MCSuperior Please follow https://github.com/IntelRealSense/realsense-ros/blob/development/README.md to install ROS.

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@MCSuperior Any other questions about this? Thanks!

MCSuperior, 2019-09-30:
I am sorry for the late reply. Due to some circumstances, I could not try the mentioned approaches yet. I will do so asap in order to see whether the approaches shown work or not.

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@MCSuperior Did you get chance to try? Any update? Thanks!

MCSuperior, 2019-10-26:
Sorry for the extended absence from this thread.
Yes, the mentioned approach worked out. Thank you for your assistance.

closed 2019-10-26 by MCSuperior