realsenseai/librealsense #4444: Intel RealSense Viewer auto-reset of parameters

AntoineWefit, 2019-07-17:
### The "Color Correction" parameters are reset when a custom ".json" is loaded
_Viewer version : 2.24_

There is a weird behavior of the "Color Correction" parameters, which are reset automatically. It's kind of difficult to know exactly what is the cause though, for example **one of the triggering action is the modification of the RGB camera parameters**. I'm guessing that there might be some incompatibilities between parameters but it's pretty annoying to see your configuration being reset every time you change something :(
Since the depth and RGB sensors are distinct, I'm not sure if this behavior is a bug or not...

[image]

dorodnic, 2019-07-20:
Hi @AntoineWefit 
Looks like a bug, we will try to reproduce and investigate

AntoineWefit, 2019-07-24:
Look a bit like [DSO-8538](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes), don't know if related though...

RealSenseSupport, 2021-03-26:
Hello @AntoineWefit , do you still find that this is happening with the latest f/w and viewer? Can you attach your json? I have not been able to reproduce this condition.
Thanks.

RealSenseCustomerSupport, 2021-08-14:
----------------------------------------------

@AntoineWefit this doesn't appear to be an issue anymore so we'll close the ticket. If you find that it still happens with the latest SDK, feel free to open a new issue.

closed 2021-08-14 by RealSenseCustomerSupport