realsenseai/librealsense #4445: rscalibrationapi DC_ERROR_TABLE_NOT_SUPPORTED

SirDifferential, 2019-07-17:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   Ubuntu 18.04 by Emutex for Upboad | 
| Kernel Version (Linux Only)    |   4.15.0-37-generic     | 
| Platform                                 | Upboard |
| SDK Version                            |  2.22                          | 
| Language                            |  C++ | 
| Segment			|  Calibration | 

### Issue Description
I'm performing the targetless calibration on an D435 using the librscalibrationapi by following https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_Programmer.pdf.

Minimal code to reproduce this can be found here: https://github.com/SirDifferential/minimal_rs_calibration

Upon calling `DynamicCalibrationAPI::DSDynamicCalibration::Initialize()`, I'm getting the following error:

```
dyncal initialize DC_ERROR_TABLE_NOT_SUPPORTED
```

Which the documentation only elaborates as:

```
DC_ERROR_TABLE_NOT_SUPPORTED - calibration coefficient table on device is not supported
```

What does this error mean and how how do I correct it to run the calibration on a D435?

Additional question: How to figure out which of the IR frames in a composite frame is for the left camera and which is for the right camera? The part of the code is around here: https://github.com/SirDifferential/minimal_rs_calibration/blob/master/main.cpp#L381

Third, do you plan on releasing a Raspberry Pi 4 compatible ARM linux version of rscalibrationapi?

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Can you run`CustomRW -r `​ to see if all the tables show up correctly?
It may be related to https://github.com/IntelRealSense/librealsense/issues/4217

SirDifferential, 2019-08-14:
This is what it says:

```
mngr@delicode-24b90eb18eeb4c:~/librscalibrationapi/deb/tool/usr/bin$ ./Intel.Realsense.CustomRW -r
CustomRW for Intel RealSense D400, Version: 2.6.8.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 825312070308
  Firmware version: 05.11.06.250

Calibration parameters from the device:
  resolutionLeftRight: 1280 800

  FocalLengthLeft: 644.130005 643.574402
  PrincipalPointLeft: 642.805481 408.183197
  DistortionLeft: -0.055474 0.063005 -0.000715 0.000266 -0.020710

  FocalLengthRight: 642.541504 641.921326
  PrincipalPointRight: 636.532166 406.338074
  DistortionRight: -0.055434 0.064647 -0.000198 -0.001628 -0.021201

  RotationLeftRight: 0.999972 0.007352 0.001340
                     -0.007350 0.999972 -0.001502
                     -0.001351 0.001492 0.999998
  TranslationLeftRight: -49.961227 0.302183 0.087509

  HasRGB: 1

  resolutionRGB: 1920 1080

  FocalLengthColor: 1387.006592 1386.820557
  PrincipalPointColor: 976.041382 554.330444
  DistortionColor: 0.000000 0.000000 0.000000 0.000000 0.000000
  RotationLeftColor: 0.999936 -0.011283 -0.001107
                     0.011280 0.999934 -0.002148
                     0.001131 0.002135 0.999997
  TranslationLeftColor: 14.916027 0.238545 0.039954
```

`DistortionColor` looks to be zero?

RealSenseCustomerSupport, 2019-08-19:
----------------------------------------------

@SirDifferential,
The calibration table looks fine and the zero RBG distortion correction is expected.
Did you try passing the rs2::device handle or RS400Device class?

    // The DSDynamicCalibration Initialize requires a rs2::device device handle. **It can be a obtained from librealsense directly or from the Rs400Device class**as below
    Rs400Device *g_rsDevice;
    void *g_pDevice;
    g_rsDevice = new Rs400Device();
    g_pDevice = g_rsDevice->GetDeviceHandle();           // get librealsense device handle from Rs400Device. It can also be obtained directly from a librealsense instance
                  //
    g_rsDevice->InitializeCamera(deviceSN)

    // Initialize Dynamic Calibration API to custom calibration mode
    DynamicCalibrationAPI::DSDynamicCalibration *m_dcApi = new DSDynamicCalibration();
    int ret = m_dcApi->Initialize(**g_pDevice**, DynamicCalibrationAPI::DSDynamicCalibration::CAL_MODE_USER_CUSTOM);

SirDifferential, 2019-08-21:
I might be missing something obvious here, but I'm not seeing a type called `Rs400Device` anywhere in librealsense (master or development branches) or in librscalibrationapi. I'm not seeing functions called `GetDeviceHandle` or `InitializeCamera` either.

The device handle I'm passing to the calibrator is the return value of `rs2_create_device(device_list, index, &errorcode)`.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Afte the librscalibrationapi is installed, can you verify the following paths/files described in section 2.2.2 and 3.3?
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_Programmer.pdf

SirDifferential, 2019-08-30:
Alright, I got this solved. The classes / functions you mentioned aren't in the librscalibrationapi headers, but in the examples under `/usr/share/doc/librscalibrationapi`. I wasn't using the examples, but read the documentation on the PDF and utilized the API directly.

The original problem with `DC_ERROR_TABLE_NOT_SUPPORTED` seems to happen if the handle you pass to `DynamicCalibrationAPI::DSDynamicCalibration::Initialize()` comes from the C API function `rs2_create_device`, while using the function `rs2::context::query_devices()` in the C++ API produces a handle that works. So this is seems to be the C API producing something librscalibrationapi doesn't understand.

I have updated the minimal calibration program to use the C++ API in the hopes someone else will find it useful. The bundled example code in librscalibrationapi is a bit too verbose and a more minimal example might be desirable.

A few recommended changes:

* Mention that you have to use the librealsense C++ API, not the C API
* Mention that you have to use the librealsense C++ API, not the C API
* You might want to edit the librscalibrationapi documentation to add a mention of safeclib being a dependency: https://github.com/rurban/safeclib
* in librscalibrationapi/usr/include/librscalibrationapi/DSOSUtils.h, this include is incorrect:

```
extern "C" {
#include "safe_mem_lib.h"
}
```

It should be: `#include <libsafec/safe_mem_lib.h>`

Now I'll just have to get the rest of the code working, but as far as I'm concerned, this issue can be closed.

RealSenseCustomerSupport, 2019-08-30:
----------------------------------------------

Hi @SirDifferential, thank you for the feedback and I am glad the issue is resolved.
We will close this ticket now.

closed 2019-08-30 by RealSenseCustomerSupport