realsenseai/librealsense #4446: The depth measuring

clear-zhu, 2019-07-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   |05.10.13.00 | 
| Operating System & Version |    Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            | librealsense2                         | 
| Language                            |  C++                       | 
| Segment			|  others                          | 

### Issue Description
My goal is to find the depth corresponding to the white pixels of the image. But the distance I found using the function get_distance() is incorrect. My box is about 0.5 meters from the camera.
The white pixels are the areas corresponding to the boxes and should all be about 0.5 meters deep.The last image is the partial coordinates and depth of the white pixels. Theoretically, the depth value should be 0.5 meters, why there are more than 30 meters of data. Before this I have matched the color image with the depth image. Thank you in advance
[image]
[image]
[image]

dorodnic, 2019-07-19:
This is rather strange. What values are you getting when hovering over the box in RealSense Viewer?
Could you possibly share the min reproduction code?

clear-zhu, 2019-07-20:
Thank you for your answer, the question does appear in the code. I've corrected my code, and now the measured depths are roughly the same

dorodnic, 2019-07-22:
Glad to hear. Closing the issue, feel free to open a follow-up issue if you have any more questions.

closed 2019-07-22 by dorodnic

manyaafonso, 2019-09-26:
Hi @clear-zhu , can you please share the corrections you made to your code to solve this?