realsenseai/librealsense #4447: clang-tidy: potential leak in 2.21.0 sensor::start

pinver, 2019-07-17:
This seems similar to issue #3873
```
       template<class T>
        void start(T callback) const
        {
            rs2_error* e = nullptr;
            rs2_start_cpp(_sensor.get(), new frame_callback<T>(std::move(callback)), &e);
            error::handle(e);
        }
```

clang-tidy reports:
```
<snip>/librealsense2/hpp/rs_sensor.hpp:190:13: warning: Potential memory leak [clang-analyzer-cplusplus.NewDeleteLeaks]
            error::handle(e);
            ^
<snip> realsense/realsense_interface.cc:123:3: note: Calling 'sensor::start'
  realsense_sensor_.start(frame_queue);
  ^
<snip>/librealsense2/hpp/rs_sensor.hpp:189:42: note: Memory is allocated
            rs2_start_cpp(_sensor.get(), new frame_callback<T>(std::move(callback)), &e);
                                         ^
<snip>/librealsense2/hpp/rs_sensor.hpp:190:13: note: Potential memory leak
            error::handle(e);
            ^
```

radfordi, 2019-07-23:
Are you saying this is an actual leak or just that `clang-tidy` thinks there is an error?  It looks to me like we initialize the `shared_ptr` with the correct destructor [immediately](https://github.com/IntelRealSense/librealsense/blob/master/src/rs.cpp#L658) and that nothing can throw in the mean time.  Am I missing something?
```
void rs2_start_cpp(const rs2_sensor* sensor, rs2_frame_callback* callback, rs2_error** error) BEGIN_API_CALL                                                                                                                                                                                                                          const rs2_context *context  context
{                                                                                                                                                                                                                                                                                                                     rs2_devices_changed_callback_ptr callback  callback
    VALIDATE_NOT_NULL(sensor);                                                                                                                                                                                                                                                                                                                           void *user  user
    VALIDATE_NOT_NULL(callback);
    sensor->sensor->start({ callback, [](rs2_frame_callback* p) { p->release(); } });
}
HANDLE_EXCEPTIONS_AND_RETURN(, sensor, callback)
```

RealSenseSupport, 2021-03-11:
Just wondering if this is still an issue with the recent SDK release. 
Thank you and the case will be closed unless otherwise.

closed 2021-04-28 by RealSenseCustomerSupport