realsenseai/librealsense #4450: Issues with T265 Camera Reporting Position

cookem4, 2019-07-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |0.0.18.5715 | 
| Operating System & Version |  Raspbian 9.9  | 
| Kernel Version (Linux Only)    |   4.19.42                                         | 
| Platform                                 | Raspberry Pi 3B+  |
| SDK Version                            |  2                    | 
| Language                            |  C                          | 
| Segment			|  Robot                          | 

### Issue Description

I have been having problems with reliability of the T265 tracking camera when used along with a Rapberry Pi 3B+. To preface this, everything works perfectly and as expected when the camera is used on an Ubuntu 16.0 computer.

The problem I am facing is the reliability of the position measurements  of the camera. When the position is streamed with the rs-pose application using the Raspberry Pi, it at random will be accurate in some cases and completely inaccurate in other cases. For example, I can lift the camera approximately 30cm in the +Y direction and it will display a position of (0,0.3,0) as expected. The result of this can be seen in the following image:
[image]

But every, say, third of the times the program is executed and the camera is lifted 30cm in the +Y direction, the reported position does not move much from the origin. The position varies by a couple centimeters as the camera is moved around but does not scale to amount the camera is being moved. The output when this occurs can be seen in the following image:

[image]

Now, none of these issues happen when the camera is used with an Nvidia Jetson TX2 running Ubuntu 16.0, this problem only occurs with a Raspberry Pi. Is this a known problem with the Raspban operating system? Are there any solutions to this problem?

radfordi, 2019-07-23:
This is not a known issue.  I don't see how the host being different would cause this.  This is likely a tracking issue.  Is the device started from the same physical location in each of your two test scenarios?

cookem4, 2019-07-23:
After testing thoroughly on the other computer that ran Ubuntu 16 the problem with the device position not updating started to pop up. The tests are also done from the same physical location each time. Do you know of any scenarios that could cause the device position to not be updated accurately?

peterwatsonnm, 2019-07-29:
I have noticed that if there is a static object in the frame of the camera (like your finger, a camera mount, robot piece, etc), it will make the device position less accurate.  Maybe double check with the viewer that there is never anything in your camera frame?

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi @cookem4 - it has been a while since this issue occurred.  Have you attempted checking the camera view to determine if there is static objects in the FOV?  Also we've posted newer releases since you encountered this issue.  Do take a look at the latest releases and see if anything has improved.

closed 2024-03-29 by MartyG-RealSense