realsenseai/librealsense #4451: Is there a way to export tracking data to a 3d software like 3d studio max and apply the tracking data to a 3d camera in the software?

frankywashere, 2019-07-17:
Is there a way to export camera track data from the T265 to a 3d software package like 3d studio max of cinema 4d and apply the tracking data to a camera in the software?

millsrandy, 2019-07-27:
I'd like to know this as well.  I was able to get the T265 ported into Unity and able to move a camera live within Unity, however, I wasn't able to record the camera position and keyframe it so that it would hold position.  I've been working on a different method by taking the CSV and stripping all the info in it, but I can't get the bag file converter that is included with librealsense to work.

frankywashere, 2019-07-27:
Interesting, I wonder what kind of programmer we’d need to figure this out. Seems the csv method should be simple.

Frank

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi @frankywashere

There are a few different examples in librealsense that might be able to help here rs-trajectory, tracking and depth (if you want to bring in a depth camera into the system).  Also the realsense-viewer has the capability of saving bag files and playback within the viewer itself.  Currently we also have APIs to export/save a map for T265 but we do not provide the specifics or details of the maps configuration.  You would only be able to load it back onto the T265 for use, not able to use a 3rd party tool to see the map.  Lastly as discussed, taking the pose data in a CSV and using a 3rd party tool to map it out.

If there is anything else you need on this topic, let us know.

Thanks

millsrandy, 2019-11-15:
@RealSenseCustomerSupport How would you propose exporting the pose data in CSV format?  The included rs-convert tool is not working currently.  Do you have any suggestions?  Thank you!

DDR83, 2020-09-25:
@RealSenseCustomerSupport hi, I am looking for a way create a CSV file with per frame position and orientation data to import in a 3rd party software. can you help me with my quest?

closed 2024-03-29 by MartyG-RealSense