realsenseai/librealsense #4452: depth measurement

GESZ24, 2019-07-17:
[image]
[image]

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | (Open RealSense Viewer  | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.18.0-15)                                         | 
| Platform                                 | PC  |
| SDK Version                           |      2.23.0.909                   | 
| Language                              |  Python                          | 
| Segment	            		|  others                          | 

### Issue Description
I´m working with depth camera D435 and I´m trying to measure short distances (mm) but i can´t change the units, I´m using the camera to mesuare the depth wheel but sometimes the depth is detected but the change that show the viewer is minimum 1 cm.

dorodnic, 2019-07-20:
Hi @GESZ24 

> but i can´t change the units

Can you explain what is exactly happening? Do you see the slider named Depth Units? Is it read-only? 
Is the camera in USB3 or USB2 mode? 
Also, under advanced controls there is another extremely handy parameter for what you are doing called Disparity Shift (under Depth Table).
Tuning Depth Units, Disparity Shift, Laser Power (to avoid over-saturation in the IR image) should help with your use case (the ruler shows ~20cm, right?) Also, different resolutions have different min-z properties.

GESZ24, 2019-07-22:
yes, thanks! it was very helpful

RealSenseCustomerSupport, 2019-08-09:
----------------------------------------------

Hi @GESZ24, the color map can be adjusted under Depth Visualization in which you can change various parameters.
We will go ahead and close this ticket if there is no further question. Thank you.

closed 2019-08-20 by RealSenseCustomerSupport