realsenseai/librealsense #4453: poll_for_frames for matlab

sshhj89, 2019-07-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435i} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {matlab }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am wondering that there is a way to call poll_for_frames like in C++ in the matlab?
Currently, the wrapped pipe does not have to function.

lramati, 2019-08-14:
There is not currently a way to call poll_for_frames.
I will implement it for inclusion in a future version of the wrapper

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@sshhj89 Below PR merged to the latest v2.34 SDK which implemented poll_for_frames for matlab. Please have a try. Thanks!

https://github.com/IntelRealSense/librealsense/pull/5970

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@sshhj89 Did you get chance to try the latest v2.34 SDK? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@sshhj89 Will close this if no other questions. THanks!

closed 2020-05-18 by RealSenseCustomerSupport