realsenseai/librealsense #4456: T265 viewer exe

deeprun, 2019-07-18:
Fro T265 on linux, is there a "viewer" exe like that on Windows?

SlavikLiman, 2019-07-18:
@deeprun ,
There is no "exe" for Linux. In order to use T265 on a Linux host please follow the following instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

deeprun, 2019-07-19:
If you install realsense on windows, there is a exe call "Viewer" which display two fisheye camera images and the device movement. I am asking if there is a similar program on linux.

hoangthien94, 2019-07-19:
I believe [`realsense-viewer`](https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer) should be the one you are referring to? 

If you have already installed Realsense SDK 2.0 following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) then you can just open a terminal and type in `realsense-viewer`

closed 2019-07-22 by dorodnic