realsenseai/librealsense #4457: D435 cameras not being detected if plugged in at boot

milanfertin, 2019-07-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.6.200 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.21.0.774                       | 
| Language                            |  C#                          | 
| Segment			|  others  | 

### Issue Description

When booting my PC with the D435 cameras already plugged in, I get a "USB not recognized error". The only working fix that I have for now is to unplug and replug the cameras. Though, this is not a viable solution for my use because my PC is being shut down in the evening and booting in the morning to be operational for the day and it is supposed to be autonomous.

The issue seems to be quite random, sometimes 1 of my 2 cameras will be detected, most of the time none of them will be, and very very rarely both will work.

My problem seems to be very similar to #4097, but sadly the issue has been closed without a fix to the initial problem.

Thank you in advance for your help.

AntoineWefit, 2019-07-18:
These cameras are indeed known for being unpredictable regarding the USB connection :(
I'm not an expert, but I would advise you to :
- update your cameras to the [latest firmware](https://downloadcenter.intel.com/download/28870/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t).
- try other USB 3.0 ports & cables (I guess that you already did that).
- maybe trigger a USB reset at boot ? I don't really know if it's going to be useful though.
- use the SDK function hardware_reset which sometimes helps, but I doubt that you can access your devices if you get a "USB not recognized" error...
- update your USB drivers.

Hope this'll help !

milanfertin, 2019-07-19:
Thank you for your answer.

I updated my cameras to the latest firmware 5.11.6.250 but the problem persists and my USB drivers are up to date as well.

I can't trigger a hardware_reset since my cameras are not detected in the first place.

I'm thinking it's about the USB ports actually, because I also have other devices plugged to my PC. So my guess is that in some cases the cameras don't receive enough power at boot and then get stuck into that error mode until I unplug/replug them. I'm gonna investigate on that today.

An input from the staff itself would be good as well, since it seems this issue is more common than what I thought.

milanfertin, 2019-07-19:
Well, I didn't find any convenient explanation to the cause of the issue. Though, it seems that using a self-powered USB hub reduces the error frequency ; I could reboot 3 times in a row without getting the error.

AntoineWefit, 2019-07-19:
It might be a power issue then, as you suggested !
Please let me know if you find a reliable fix, because I'm also interested.

milanfertin, 2019-07-19:
Alright, I have been doing some more tests and making up crazy theories (at this point nothing seems crazy anymore, though).

So my PC has in total 4 USB 3.0 ports. I noticed when doing some tests that when only the 2 cameras are plugged in, the error happens often.

Since the software that I work on and that requires the cameras is a touchscreen app, I don't need a keyboard or a mouse. Though, to easily access Windows device manager and other stuff to check if cameras are detected at boot I plugged 1 USB mouse or 1 USB keyboard. Though, I once randomly plugged both the keyboard and the mouse, and then the cameras started to be detected everytime. 

So I thought: "The built-in USB root hub of my PC might allocate an amount of power based on how many devices are plugged in". Since my app also requires 2 other devices (usb touchscreen and usb scale), I replaced my mouse and keyboard by these 2 other devices and restarted my computer. Though, it didn't detect the cameras every time anymore.

So I thought that maybe it was just lucky randomness that happened with my mouse and keyboard. But when I plugged them again instead of the 2 other devices it started working again.

So my theory is that my root hub gets an amount of power based on how many devices are plugged to it. But if the other devices are too demanding then the cameras don't get enough for themselves and then the error appear.

Could you maybe try something similar if you experience the same issue?

milanfertin, 2019-07-19:
If others don't get the same problem and if the issue is actually power-related I guess it all depends on how the USB hub handles the power distribution to all plugged devices. So there would not  be much Intel can do about it except reduce the required power for their cameras. But that would probably cause a loss of quality.

I'm gonna do some more tests beginning of the next week and try it out on production if it all goes right.

I will of course keep you informed on the status! Could also be useful for future D435 users that encounter the same problem. Thanks again for your answers and have a nice weekend!

milanfertin, 2019-07-25:
Hello @AntoineWefit,

Coming back to you a little bit later than expected because I had other stuff to focus on at work this week.

So, after doing some more testing it seems that my theory was right. Though, I'm unsure if it's power related or if it's due to data transfer capacity.

I hope for other users reading this and looking for a solution that it's not only specific to my case and that they will understand how to fix the issue after reading this because it took me a lot of time to understand this weird behavior.

AntoineWefit, 2019-07-25:
> So, after doing some more testing it seems that my theory was right.

Using a USB hub with a dedicated power supply would be a fix then ?

milanfertin, 2019-07-26:
I encountered issues if I plugged only 2 cameras on a USB hub with a dedicated power supply. Same behavior as if I used my PC's root hub. 

That's why I'm unsure if it's power related or if there is some kind of data transfer capacity being exceeded.

m9dfukc, 2019-08-13:
hey,

we experience the exact same issue. our setup are 3x D435 cameras connected to a powered usb3 hub. The behaviour seems to be random, sometimes all three cameras work sometimes the second and the third camera are not being detected after the boot process.

milanfertin, 2019-08-13:
Hello,

Since I solved my problem I didn't spend more time trying to find a real fix. But you should check [this comment](https://github.com/IntelRealSense/librealsense/issues/4552#issuecomment-518370718). There is a chance it could fix our issue.

m9dfukc, 2019-08-13:
@milanfertin 
thanks for the link! will definitely try this out!!

closed 2019-09-05 by RealSenseCustomerSupport