realsenseai/librealsense #4459: Random Segmentation Fault when capturing images

pierreavanzini, 2019-07-18:
----------------------------------------------------------------------------------------------------

| Required Info                          |                                                       |
|---------------------------------------|------------------------------------------- |
| Camera Model                        |  D435
| Firmware Version                    |  05.09.02.00
| Operating System & Version   | Ubuntu 18.04.2 LTS
| Kernel Version (Linux Only)    |  4.15.0-37-generic x86_64 
| Platform                                  |  intel UP board
| Language                               |  C++ / opencv
| Segment			         |  Robot / Drone                               | 

### Issue Description
With an UP board and a USB3 hub, I capture color and depth images from four D435 cameras and my algorithm ends randomly with a segmentation fault (If I only use one camera I have the same problem but the backtrace is a bit different).

Backtrace with 4 devices:

Thread 106 "RealSense" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffaaffd700 (LWP 2730)]
0x00007ffff70303d7 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x7fff70069140) at /usr/include/c++/7/bits/shared_ptr_base.h:154
154		    _M_dispose();
(gdb) bt
#0  0x00007ffff70303d7 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x7fff70069140) at /usr/include/c++/7/bits/shared_ptr_base.h:154
#1  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::operator= (__r=..., this=0x7ffff7d21fa8 <librealsense::environment::get_instance()::env+200>) at /usr/include/c++/7/bits/shared_ptr_base.h:703
#2  std::__shared_ptr<librealsense::platform::time_service, (__gnu_cxx::_Lock_policy)2>::operator= (this=0x7ffff7d21fa0 <librealsense::environment::get_instance()::env+192>)
    at /usr/include/c++/7/bits/shared_ptr_base.h:1034
#3  std::shared_ptr<librealsense::platform::time_service>::operator= (this=0x7ffff7d21fa0 <librealsense::environment::get_instance()::env+192>) at /usr/include/c++/7/bits/shared_ptr.h:93
#4  librealsense::environment::set_time_service (this=this@entry=0x7ffff7d21ee0 <librealsense::environment::get_instance()::env>, ts=
    std::shared_ptr<librealsense::platform::time_service> (use count 2, weak count 0) = {...}) at /home/up/librealsense/src/environment.cpp:201
#5  0x00007ffff702080d in librealsense::context::context (this=0x7fff7c008d30, type=<optimized out>, filename=<optimized out>, section=0x0, mode=RS2_RECORDING_MODE_COUNT, min_api_version="0.0.0")
    at /home/up/librealsense/src/context.cpp:131
#6  0x00007ffff706400f in __gnu_cxx::new_allocator<librealsense::context>::construct<librealsense::context, librealsense::backend_type> (this=<optimized out>, __p=0x7fff7c008d30)
    at /usr/include/c++/7/ext/new_allocator.h:136
#7  std::allocator_traits<std::allocator<librealsense::context> >::construct<librealsense::context, librealsense::backend_type> (__a=..., __p=<optimized out>)
    at /usr/include/c++/7/bits/alloc_traits.h:475
#8  std::_Sp_counted_ptr_inplace<librealsense::context, std::allocator<librealsense::context>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<librealsense::backend_type> (__a=..., 
    this=0x7fff7c008d20) at /usr/include/c++/7/bits/shared_ptr_base.h:526
#9  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<librealsense::context, std::allocator<librealsense::context>, librealsense::backend_type> (__a=..., this=0x7fff7c068e78)
    at /usr/include/c++/7/bits/shared_ptr_base.h:637
#10 std::__shared_ptr<librealsense::context, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<librealsense::context>, librealsense::backend_type> (__a=..., __tag=..., this=0x7fff7c068e70)
    at /usr/include/c++/7/bits/shared_ptr_base.h:1295
#11 std::shared_ptr<librealsense::context>::shared_ptr<std::allocator<librealsense::context>, librealsense::backend_type> (__a=..., __tag=..., this=0x7fff7c068e70)
    at /usr/include/c++/7/bits/shared_ptr.h:344
#12 std::allocate_shared<librealsense::context, std::allocator<librealsense::context>, librealsense::backend_type> (__a=...) at /usr/include/c++/7/bits/shared_ptr.h:691
#13 rs2_create_context (api_version=<optimized out>, error=0x7fffaaffcc08) at /usr/include/c++/7/bits/shared_ptr.h:707
#14 0x000055555558aaad in rs2::context::context() ()
#15 0x000055555558c0e6 in CamInfo::CamInfo() ()
#16 0x000055555558c348 in FrameInfo::FrameInfo() ()
#17 0x00005555555871d1 in capture_thread(CamInfo) ()
#18 0x000055555558f40d in void std::__invoke_impl<void, void (*)(CamInfo), CamInfo>(std::__invoke_other, void (*&&)(CamInfo), CamInfo&&) ()
#19 0x000055555558d5fe in std::__invoke_result<void (*)(CamInfo), CamInfo>::type std::__invoke<void (*)(CamInfo), CamInfo>(void (*&&)(CamInfo), CamInfo&&) ()
#20 0x000055555559755b in decltype (__invoke((_S_declval<0ul>)(), (_S_declval<1ul>)())) std::thread::_Invoker<std::tuple<void (*)(CamInfo), CamInfo> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) ()
#21 0x0000555555597470 in std::thread::_Invoker<std::tuple<void (*)(CamInfo), CamInfo> >::operator()() ()
#22 0x0000555555596816 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(CamInfo), CamInfo> > >::_M_run() ()
#23 0x00007ffff4b4b66f in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#24 0x00007ffff42c06db in start_thread (arg=0x7fffaaffd700) at pthread_create.c:463
#25 0x00007ffff3fe988f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Backtrace with only 

one device:

Thread 18 "RealSense" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe57f2700 (LWP 16887)]
0x00007fffcc006c00 in ?? ()
(gdb) bt
#0  0x00007fffcc006c00 in ?? ()
#1  0x000055555558dc50 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#2  0x00007ffff707d255 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7fffe57f15e8, __in_chrg=<optimized out>) at /usr/include/c++/7/bits/shared_ptr_base.h:684
#3  std::__shared_ptr<librealsense::platform::time_service, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7fffe57f15e0, __in_chrg=<optimized out>) at /usr/include/c++/7/bits/shared_ptr_base.h:1123
[... 272 lines cut ...]
}

void intrinsics_to_csv(const rs2::pipeline_profile& profile, const std::string& filename)
{
    rs2::depth_sensor sensor = profile.get_device().first<rs2::depth_sensor>();
    rs2::video_stream_profile stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
    rs2_intrinsics intrinsics = stream.get_intrinsics();
    rs2_distortion model = intrinsics.model;
    
    std::ofstream csv;
    csv.open(filename);
    csv << "Image Width," << stream.width() << "\n";
    csv << "Image Height," << stream.height() << "\n";
    csv << "Principal Point X," << std::setprecision(10) << intrinsics.ppx << "\n";
    csv << "Principal Point Y," << std::setprecision(10) << intrinsics.ppy << "\n";
    csv << "Focal length X," << std::setprecision(10) << intrinsics.fx << "\n";
    csv << "Focal length Y," << std::setprecision(10) << intrinsics.fy << "\n";
    csv << "Distorsion Model," << model << "\n";
    for (int i=0; i<5; i++)
    {
        csv << "Coef " << i <<","<< std::setprecision(10) << intrinsics.coeffs[i] << "\n";
    }
    csv << "Depth Scale," << std::setprecision(10) << sensor.get_depth_scale() << "\n";
    csv.close();
}

pierreavanzini, 2019-07-22:
No one to help me ?

pierreavanzini, 2019-07-23:
When I run the algorithm I also often have this error:

./RealSense 
Configuring camera : 817412070341
Starting pipeline for current camera 817412070341
Configuring camera : 819312073507
Starting pipeline for current camera 819312073507
Configuring camera : 815412070067
Starting pipeline for current camera 815412070067
Configuring camera : 817512070711
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55818fc09db0, config:0x55818f4c96f0):
    hwmon command 0x10 failed. Error type:  (1).

The error can occur on any camera (not one in particular). For information I use a powered USB3 hub.
Thank you.

RealSenseCustomerSupport, 2019-08-14:
----------------------------------------------

Hi @pierreavanzini,

It looks the basic query of the device version and info has some trouble per below log.
1. Do you try your camera in another Ubuntu PC instead of UP board?
2. Is there any trouble to run the librealsense apps like rs-capture, rs-enumerate-devices, etc ?

    hwmon command 0x10 failed. Error type: (1).

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @pierreavanzini,

Any update? Is the issue clarified/resolved/

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Hi @pierreavanzini,

closed 2019-12-05 by RealSenseCustomerSupport