realsenseai/librealsense #4460: D435 not recognized by RealSense Viewer or Matlab (Win 10)

SgtJaap, 2019-07-18:
I can't seem to get my **D435** to work on **Win 10** on my Macbook (with i7 4870HQ and USB3). I have the latest SDK 2.0 build (2.24.0) and have also tried previous builds. Whatever I try, the RealSense Viewer does not detect the D435. It just says "connect a RealSense Camera or Add Source" in the top left corner. Like this:
[image]
Matlab, where I want to work with the camera, also does not detect it. This camera works on my friends Win 10 laptop with both the RealSense viewer and Matlab, but I really need it to work on mine too. (To work on for uni)

The camera does get recognized by Windows in Device Manager but as a USB device, not a camera. 
[image]

What could be the issue and what could possibly solve this?

dorodnic, 2019-07-19:
In the settings window of the Viewer you can enable logging to file. Please attach the log when connecting the camera.
I suspect there is some internal Windows error due to OS privacy policies, but knowing the exact error code will help us troubleshoot the problem.

SgtJaap, 2019-07-19:
> In the settings window of the Viewer you can enable logging to file. Please attach the log when connecting the camera.
> I suspect there is some internal Windows error due to OS privacy policies, but knowing the exact error code will help us troubleshoot the problem.

I tried multiple times but the log file is always empty.

ev-mp, 2019-08-05:
@SgtJaap , did you check that the `camera privacy setting` are configured properly ? #4259, #3381

RealSenseCustomerSupport, 2019-09-17:
----------------------------------------------

@SgtJaap With the suggestion from ev-mp, did you still have issue? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@SgtJaap  Will close this if no other questions. Thanks!

closed 2019-09-29 by RealSenseCustomerSupport