realsenseai/librealsense #4461: Report a bug, "foreach", and solution

xzlxiao, 2019-07-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    MacOS  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.24.0                        | 
| Language                            |  c++                      | 
| Segment			|                          | 

### Issue Description
report a bug:
When I used this sdk with qt, a problem would be encountered.
A function called foreach in file "rs_frame.hpp" was in conflict with qt.
I renamed the function to get through compilation.
I suggest you will change the name of this function in the source code.

dorodnic, 2019-07-20:
Hi @xzlxiao 
Yes, you are right. There was a PR (#2470) trying to address this exact same issue. The original pull-request was not passing CI (it also requires small fixes in the python wrapper), and I forgot to fix and merge it myself. Should be done.

dorodnic, 2019-11-11:
Should be resolved today

closed 2019-11-19 by dorodnic