realsenseai/librealsense #4462: Realsense Viwer does not even start in Windows 7 8 or 10

dbarry-mitek, 2019-07-18:
Very simple   - I download the Realsense viewer .exe onto several machines and it just does not even start, no errors.

The instructions say ->

Step 2: Install the Intel® RealSense™ SDK and run the Intel® RealSense™ viewer

The Intel® RealSense™ viewer provides immediate access to your depth camera data streams.

For Windows* users:

Go to the latest release page on GitHub*.
Download Intel.RealSense.Viewer.Portable.exe
Double-click the executable file to run the program.

There is no file named   Intel.RealSense.Viewer.Portable.exe  only
Intel.RealSense.Viewer.exe

dbarry-mitek, 2019-07-18:
> The correct instructions for Windows installation are at the link below.
>
> https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_windows.md

That is what I have done, and it didn’t work.   Interesting that the link you provided shows different file sizes.

[cid:image005.jpg@01D53D76.D18E5920]

[cid:image006.jpg@01D53D76.D18E5920]

MartyG-RealSense, 2019-07-18:
I can't see the images that you are referring to as the GitHub is not displaying them.  If you mean this list:

Depth.Quality.Tool.exe   9.45 mb
Intel.RealSense.SDK.exe  129 mb
Intel.RealSense.Viewer.exe   7.31 mb

Then the full SDK download is always the largest of the files shown, the one called 'Intel.RealSense.SDK' plus a version number on the end.   For example, for the Windows 10 version:

**Intel.RealSense.SDK-WIN10-2.24.0.939.exe**

And for the Windows 7 version:

**Intel.RealSense.SDK-WIN7-2.24.0.939.exe**

closed 2019-07-22 by dorodnic