realsenseai/librealsense #4463: Issue when closing the camera sensor

Samvid95, 2019-07-18:
How to check whether the sensor is streaming or not? 

So I am trying to close out all the sensor to clean up the code. But I am getting the error that `sensor `is not streaming so I cannot close the `sensor`. So I am looking on how to check that. 

Also what do I need to do in order to close the D435? I am doing `pipe.stop()` is it enough or do I need to close the sensors as well?

dorodnic, 2019-07-22:
Sensor / Pipeline destructors handle all the clean-up for you, assuming you are using C++.

RealSenseCustomerSupport, 2019-08-09:
----------------------------------------------

Hi Samvid95,
Do you still have question about this?

Samvid95, 2019-08-09:
Ohh yess. It is now resolved. I am good. You can close the issue.

closed 2019-08-09 by Samvid95