realsenseai/librealsense #4464: Ubuntu HWE kernel patch failure

themikem, 2019-07-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    | 4.18                                          | 
| Platform                                 | PC  |
| SDK Version                            | N/A                          | 
| Language                            | N/A                          | 
| Segment			|  N/A                          | 

### Issue Description
Looks like the Ubuntu HWE branch has moved to kernel 5.0, but the patch script is still trying to patch 4.18 files.

A fresh checkout from librealsense yeilds:
```
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-hwe folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Cloning into './ubuntu-bionic-hwe'...
remote: Counting objects: 67842, done.
remote: Compressing objects: 100% (64251/64251), done.
remote: Total 67842 (delta 5012), reused 40605 (delta 2564)
Receiving objects: 100% (67842/67842), 199.16 MiB | 4.57 MiB/s, done.
Resolving deltas: 100% (5012/5012), done.
Checking out files: 100% (64426/64426), done.
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 219 with fuzz 2 (offset 10 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 165 with fuzz 1 (offset 7 lines).
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1302 (offset 51 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 680.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
```

However, if I manually clone the repo into /ubuntu-bionic-hwe like so:
git clone --branch Ubuntu-hwe-4.18.0-24.25_18.04.1 git://kernel.ubuntu.com/ubuntu/ubuntu-bionic.git --depth 1 ./ubuntu-bionic-hwe

and then run ./scripts/patch-realsense-ubuntu-lts.sh, the script completes successfully.

dorodnic, 2019-07-19:
Hi @themikem 
Thanks for reporting, we will look into it and update the relevant scripts

themikem, 2019-07-19:
@dorodnic

Okay great, thanks!

One observation from manually getting this to build - the HEAD on the hwe branch is pointed at a 5.0 version, but none of my LTS machines have seen an update beyond 4.18 from the Ubuntu repos. My suspicion is that there is a lag between the package release in the kernel git hosted by Ubuntu and distribution to the official repos. 

So for what it’s worth, it may make sense to convert your script to clone a tag rather than just the hwe branch proper, especially since you’re already pulling depth 1 anyway. Might prevent this issue from re-occurring next time there’s a gap like this between deb and git.

But like I said, just a thought. There are probably many other things I’m not thinking about...

Thanks again. Appreciate the responsiveness of the realsense team to issues like these.

themikem, 2019-07-25:
@dorodnic Perhaps this might help: https://github.com/IntelRealSense/librealsense/pull/4509

boilerbots, 2019-08-01:
HWE is now updating to 5.0 kernels, my machines just did and librealsense  totally fails.

What is the plan on landing these changes permanently in the kernel's main branch so we don't have to keep up with these kernel module patches?

oveddan, 2019-08-08:
thank you @themikem that manual clone solved the issue I had here!

themikem, 2019-08-08:
@oveddan You're quite welcome.  This might be easier than patching manually in the future though: https://github.com/IntelRealSense/librealsense/pull/4509 .  Let me know if it works for you.

dorodnic, 2019-08-09:
@boilerbots - yes. Over last couple of years we have upstreamed format support for all types of RealSense device and took active part in per-frame-metadata patches getting merged. 
There are only couple of features missing from latest kernels that we work to upstream as well, but if you are familiar with the process you know it takes time.
We also invest in user space drivers that do not require kernel patching (somewhat similar to libfreenect) please see libuvc_installation.md

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ev-mp, 2019-09-19:
Was [resolved](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#bug-fixes)  in v2.28
Similar to  #4624

closed 2019-09-19 by ev-mp