realsenseai/librealsense #4465: D435i on Raspberry Pi 3B. No RGB image or IMU data.

hanley6, 2019-07-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.1.100 | 
| Operating System & Version |   Ubuntu MATE 18.04  | 
| Platform                                 | Raspberry Pi 3B  |
| SDK Version                            | 2.23.0                     | 

Hi. I can follow the Realsense for Raspberry Pi instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md) and I can receive IR and Depth images in the RealSense Viewer and via ROS topics. That's great, but I'd really like to get the RGB image and IMU from the camera. I'm curious, should I have expected not to receive RGB or IMU data using the Raspberry Pi (maybe due to a lack of USB 3 ports)?

dorodnic, 2019-07-19:
This is due to the camera being connected via USB2. I would strongly recommend using a board with USB3, since USB2 will always limit the capabilities of the camera.
IMU and RGB for D435i in USB2 was a requested feature, it was implemented in firmware and either released or about to be released in upcoming firmware version. Please try updating the firmware.
Note that updating firmware using USB2 is also not recommended (may or may not work), but you can just plug the camera to any Windows / Linux PC or Mac and update directly from the Viewer.

RealSenseCustomerSupport, 2019-08-07:
----------------------------------------------

Hi hanley6,
Do you have more question about this?

closed 2019-08-07 by hanley6