realsenseai/librealsense #4466: Can't access any blob data with Unity, Realsense 2.0 sdk  and Rs415 camera

kingbaggot, 2019-07-19:
| Camera Model                       | {  D415 } | 
| Firmware Version                   | (05.09.02.00) | 
| Operating System & Version |   {Win (10) /  | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.0 }  | 
| Language                            |  {unity } | 
| Segment			|  {?? }  | 

### Issue Description
Can't access any blob data from within Unity wrapper - am looking to detect touch points on a projected-on-from-above table.  
Have talked to developers who have achieved this using older cameras/sdk combos using realSenseSettings.BlobDataOutput.QueryNumberOfBlobs() & associated functions.

If this is no longer possible - what are my options ? Really want to track those (+6) hands

MartyG-RealSense, 2019-07-19:
I used hand tracking extensively in the 2016 SDK's Unity implementation, and trialed blob tracking as part of that process.  It worked in tracking the palm but it was not too useful in terms of practical use, as you had to have your hand a few inches from the camera in order for tracking to activate (compared to hand joint tracking, which would work from about 3/4 of a meter away from the camera).

I have seen questions from a couple of different people this week about developing a projection table with overhead projectors and cameras.  Normally, the Nuitrack SDK would be a good option for tracking joints with the 400 Series cameras, but Nuitrack cannot process the joints when the camera is vertically oriented instead of facing horizontally forward, preventing this approach from being used.

In the old SDK, I was able to track the joints of my hand from above (face landmark tracking did not work from above though).  This principle is demonstrated in the HP Sprout.  It was a desktop computer in the original RealSense generation that had a RealSense camera mounted pointing downward at a flatbed scanning surface.  The camera used inside it was the F200, which was the original RealSense camera and the predecessor of the SR300 (both of which can detect hand joints).

https://www.youtube.com/watch?v=H1d1KhFFCUw

The new RealSense SR305 camera model launching this month lists finger tracking in its feature list, just like its SR300 ancestor had. 

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-sr305.html

Since the SR305 is compatible with the current RealSense SDK 2.0 but apparently not the 2016 SDK, I'm not sure what Intel's near-future plans for making use of the SR305's finger tracking functionality would be.   The RealSense SDK Manager @dorodnic may be able to offer some advice about that.    

Anyway, the OpenCV vision software has plenty of support for blob tracking (google 'opencv blob tracking') and since RealSense SDK 2.0 fully supports Unity, that may provide an avenue for you to add blob tracking to a D415-powered project.

kingbaggot, 2019-07-22:
ok cheers !

closed 2019-07-22 by kingbaggot