realsenseai/librealsense #4468: Filtering and point cloud calculation are slow on up-board

tarekmuallim, 2019-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   |       | 
| Operating System & Version |   Ubuntu 16.04.6 LTS  | 
| Kernel Version (Linux Only)    |                                            | 
| Platform                                 | up-board  |
| SDK Version                            |                          | 
| Language                            |  C++                           | 
| Segment			|  others                           | 

I am trying to use realsense D415 and up board for a real time application.
The problem that the spatial filter provided by librealsense somehow slow.
I tried to measure the time needed for the calculation and I found that:
The spatial filter takes almost 200ms using up board while it takes 30ms on my computer with i7 processor.
Also the point cloud calculation takes long time. On up board it takes 90ms while it takes 12ms on my computer with i7 processor.
Is there any way to accelerate the calculation? For example, implement the gpu for filter and point cloud calculations.

dorodnic, 2019-07-20:
Hi @tarekmuallim 
First, if you are building from source, please make sure you have `-DCMAKE_BUILD_TYPE=Release` in your cmake line.
Next, you might want to check out `rs-gl` example. It's not necessarily faster, but can offer an alternative you could leverage. 
Regarding the filters, from my experience spatial filter is the one taking the most time and giving least quality improvement, so you might decide to drop it. 
In parallel, we continue to optimize all kinds of SDK processing for all kinds of platforms (SSE4 for the Up-Board, AVX2 for newer Intel CPUs, CUDA for the Jetson family, GLSL for non-NVidia GPUs, etc...)
It's quite a large matrix and these activities are happening in parallel to new features and bug-fixes, so it takes some time.

tarekmuallim, 2019-07-22:
Thank you for the reply.
My application is for dimensions measurement purposes. And the filters is important, it improves the results.
Right now I'm trying to implement my own filters. 
Since most of the calculations can be done in parallel, you may consider using the GPU. 
I think OpenCL could be a good choice to do this. And the same code can work for Intel GPUs or Nvidia GPUs.

dorodnic, 2019-07-22:
My concern with OpenCL is its deployment on various Linux systems, but it is an option.

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@tarekmuallim Any other questions about this? Looking forward to your update. Thanks!

tarekmuallim, 2019-09-20:
No other questions. 
Thank you for your support.

closed 2019-09-23 by RealSenseCustomerSupport