realsenseai/librealsense #4469: Segfault in align with a debug build

knicos, 2019-07-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.15.0-54  | 
| Platform                                 | PC  |
| SDK Version                            |  2.24.0                        | 
| Language                            |  C++                          | 
| Segment			|  AR                          | 

### Issue Description
When using a debug build with -D_DEBUG, align frames to depth (or colour) results in a segmentation fault whilst performing an atomic operation inside a shared pointer of a stream profile. The relevant file is rs_frames.hpp line 376 (`profile = other.profile;`). In release mode there are no problems.

RealSenseCustomerSupport, 2019-09-19:
----------------------------------------------

@knicos I tried the latest SDK v2.28 and FW 5.11.11.100 and didn't get the issue. Could you please try the latest SDK and FW to see if any luck? Thanks!

RealSenseCustomerSupport, 2019-09-26:
----------------------------------------------

@knicos Could you please update if you still get issue with the latest SDK and FW? Thanks!

knicos, 2019-09-26:
I updated to 2.28.1 and the latest FW update but I do still have the segfault in the debug build. It does not appear to be resolved for me.

ev-mp, 2019-09-26:
@knicos , is this reproducible with the SDK tools/examples, such as `rs-align` ?
If not -  can you share the reproduction snippet?

knicos, 2019-09-26:
The examples do work correctly. We are using it as a part of a much larger application that has a lot of threading with no guarantee that it is the same thread calling align.process for each frame. The application also uses OpenCV and CUDA extensively in case this matters. Unfortunately I don't have time to construct an example to reproduce it and can't provide our full source.

In a class constructor:
```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGRA8, 30);

rs2::pipeline_profile profile = pipe_.start(cfg);
rs2::device dev = profile.get_device();
rs2_intrinsics intrin = profile.get_stream(rs2_stream::RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics();

rs2::depth_sensor ds = dev.query_sensors().front().as<rs2::depth_sensor>();
scale_ = ds.get_depth_scale();
```

And in a process frame method:

```
rs2::frameset frames;
if (!pipe_.poll_for_frames(&frames)) return false;
frames = align_to_depth_.process(frames);
```

The last line is the problem in debug mode.

knicos, 2019-09-26:
Ok, please ignore my previous comment. The examples DO NOT WORK if -D_DEBUG is passed to compiler. `rs-align` then segfaults.

ThomasBrandonD, 2019-10-23:
Did you append a cmake generated makefile to add the -D_DEBUG flag? I want to reproduce the segfault you're seeing

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@knicos If you're thinking to build debug version, -D_DEBUG flag is not the right one. I think you should build with **-DCMAKE_BUILD_TYPE=Debug  **
Could you please have a try and update? Thanks!

knicos, 2019-11-01:
I do use -DCMAKE_BUILD_TYPE=Debug, and that alone seems ok. However, other code we have, in libraries that are not our own, uses _DEBUG and we need to enable it.

whatisor, 2019-11-15:
#2567

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@knicos Could you please try gdb to get the core dump information about the Segfault? Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@knicos Did you get chance to try gdb to get the core dump information? Looking forward to your update. Thanks!

whatisor, 2020-02-14:
@RealSenseCustomerSupport Have you checked my gdb log or try to combine Realsense with OpenVINO on Raspberry Pi 3/4?
I am still stuck with the Segmentation fault problem of the Alignment API. As a result, I cannot use for  Intel device for my project.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@whatisor Your issue seems closed on github. Please create another new issue if it's not resolved. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@knicos  Any update from your side? Thanks!

knicos, 2020-02-17:
Any idea how to compile rs-align? Don't remember what I did last time but 2.31.1 doesn't compile on my machine, it fails to find libusb (it is installed), attempts to compile an internal version which also fails (missing config.h). I can't spend time on this so if it doesn't work I can't help further.

I can give you this from our own application which still crashes in debug (with _DEBUG) but the error is different (stack smashing detected).

#0  0x00007fffda6f1e97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fffda6f3801 in __GI_abort () at abort.c:79
#2  0x00007fffda73c897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fffda869988 "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007fffda7e7cd1 in __GI___fortify_fail_abort (need_backtrace=need_backtrace@entry=false, msg=msg@entry=0x7fffda869966 "stack smashing detected") at fortify_fail.c:33
#4  0x00007fffda7e7c92 in __stack_chk_fail () at stack_chk_fail.c:29
#5  0x00007ffff5ab8608 in librealsense::stream_filter_processing_block::should_process(rs2::frame const&) (this=<optimised out>, frame=...) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/proc/synthetic-stream.cpp:308
#6  0x00007ffff5ab94dc in librealsense::generic_processing_block::<lambda(rs2::frame, const rs2::frame_source&)>::operator() (source=..., f=..., __closure=0x55555712bc78)
    at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/proc/synthetic-stream.cpp:69
#7  0x00007ffff5ab94dc in rs2::frame_processor_callback<librealsense::generic_processing_block::generic_processing_block(char const*)::<lambda(rs2::frame, const rs2::frame_source&)> >::on_frame(rs2_frame *, rs2_source *) (this=0x55555712bc70, f=<optimised out>, source=<optimised out>)
    at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/obj-x86_64-linux-gnu/../include/librealsense2/hpp/rs_processing.hpp:128
#8  0x00007ffff5ab60a7 in librealsense::processing_block::invoke(librealsense::frame_holder) (this=0x555556853f80, f=...) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/proc/synthetic-stream.cpp:43
#9  0x00007ffff5c223f1 in librealsense::synthetic_sensor::<lambda(librealsense::frame_holder)>::operator() (__closure=<optimised out>, f=...) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/sensor.cpp:1473
#10 0x00007ffff5c223f1 in librealsense::internal_frame_callback<librealsense::synthetic_sensor::start(librealsense::frame_callback_ptr)::<lambda(librealsense::frame_holder)> >::on_frame(rs2_frame *) (this=<optimised out>, fref=<optimised out>) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/types.h:935
#11 0x00007ffff5c3331b in librealsense::frame_source::invoke_callback(librealsense::frame_holder) const (this=0x555557095bc0, frame=...) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/source.cpp:125
#12 0x00007ffff5c187bb in librealsense::uvc_sensor::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)>::operator()(librealsense::platform::frame_object, std::function<void()>) (__closure=<optimised out>, f=..., continuation=..., p=...)
    at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/sensor.cpp:377
#13 0x00007ffff5c18f27 in std::_Function_handler<void(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>), librealsense::uvc_sensor::open(const stream_profiles&):---Type <return> to continue, or q <return> to quit---
td::_Any_data &, librealsense::platform::stream_profile &&, librealsense::platform::frame_object &&, std::function<void()> &&) (__functor=..., __args#0=..., __args#1=..., __args#2=...)
    at /usr/include/c++/7/bits/std_function.h:316
#14 0x00007ffff5b437ee in std::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) const (__args#2=..., __args#1=..., __args#0=..., this=0x555557104890) at /usr/include/c++/7/bits/std_function.h:706
#15 0x00007ffff5b437ee in librealsense::platform::v4l_uvc_device::poll() (this=this@entry=0x555557104730) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/linux/backend-v4l2.cpp:954
#16 0x00007ffff5b44058 in librealsense::platform::v4l_uvc_device::capture_loop() (this=0x555557104730) at /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/linux/backend-v4l2.cpp:1290
#17 0x00007fffdb11766f in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007ffff6ddb6db in start_thread (arg=0x7fff2cff9700) at pthread_create.c:463
#19 0x00007fffda7d488f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

knicos, 2020-02-17:
Just taking another look. Is this not the problem?

Lines 623 in rs_frame.hpp (release 2.31.1)
```
#ifdef _DEBUG
        stream_profile profile;
        unsigned long long frame_number = 0;
#endif
```

If your debug build is without _DEBUG but I have _DEBUG then the rs_frame class is of a different size and hence will corrupt the stack as I'm seeing.

RealSenseCustomerSupport, 2020-02-18:
----------------------------------------------

@knicos  Did you install "libusb-1.0-0-dev" for libusb? And did you build SDK from source code or use the DKMS binary? What's CPU of your setup?

ThomasBrandonD, 2020-02-19:
> @RealSenseCustomerSupport Have you checked my gdb log or try to combine Realsense with OpenVINO on Raspberry Pi 3/4?
> I am still stuck with the Segmentation fault problem of the Alignment API. As a result, I cannot use for Intel device for my project.

@whatisor have you tried wrapping the C API in your C++ program? I've run into the same segfaults everyone else has but I've found success in wrapping the C API align functions in my C++ application

whatisor, 2020-02-19:
@ThomasBrandonD thanks for response.
I think Realsense is C++, isn't it?
I used rs2::align

knicos, 2020-02-19:
1) ... don't know (can't check atm, probably the wrong one)
2) No, always the DKMS binary.
3) Intel Core i7

I was adding _DEBUG to my own cmake for my own purposes, not rebuilding realsense SDK with it. Hence the conflict.

ThomasBrandonD, 2020-02-19:
@whatisor 

It is, but you can still use C in your C++ app, check out https://stackoverflow.com/questions/1041866/what-is-the-effect-of-extern-c-in-c

I'm assuming you're using some API to modify your the image frames you're getting from the realsense (like OpenCV). You can just use the Realsense C API to get the data into a cv::Mat

If that makes sense give it a shot or I can make a small example

whatisor, 2020-02-20:
@ThomasBrandonD , Do you mean issue only occurs for C++ API of realsense?

ThomasBrandonD, 2020-02-20:
@whatisfor Yep, it doesn't exactly "solve" the issue of this ticket but if you need a solution, using the C API is one work-around you might try which worked for me

closed 2024-03-29 by MartyG-RealSense