realsenseai/librealsense #4471: Building pyrealsense2 Visual Studio Errors

atlee19, 2019-07-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {python 3.7 }                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi guys I'm trying to build from source so I can get a wrapper for python 3.7.4 (64-bit). I got cmake to build the python bindings and create the librealsense.sln. But when I go to build pyrealsense2 in Visual Studio I'm getting 13 errors and 184 warnings. I'm also not seeing where the realsense2.dll and pyrealsense2.pyd are supposed to be saved. But i'd imagine that's because of the VS build errors. If anyone has any ideas or could help it would be much appreciated. I'll include a screenshot of the VS error log. Thanks 

[image]

dorodnic, 2019-07-22:
Hi @atlee19 
You need to resolve missing ATL includes on your system or use pre-built version of the wrapper (either one distributed in the SDK download, or via pip)

atlee19, 2019-07-22:
Hi @dorodnic 
Thanks for responding. Do you know if the pre-built version of the wrapper supports the latest version of Python(3.7.4) and 64 bit?

atlee19, 2019-07-23:
Hi, disregard my last inquiry. It turns how I had to install ATL support for C++ in Visual Studio. That's where the missing files were coming from. After that, I was successfully able to build the python wrapper from the source.

closed 2019-07-23 by atlee19