realsenseai/librealsense #4473: Some details about intel D415 camera

teriterance, 2019-07-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Operating System & Version |   Linux (Ubuntu 14/16/18)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| Language                            |  {C }                          | 

### Issue Description
I want to know if it is possible to insert my program inside Intel D415 camera, or use then processor to compute a part of my program.

dorodnic, 2019-07-22:
Hi @teriterance 
No, the D415 is fixed-pipeline ASIC.

teriterance, 2019-07-22:
hi @dorodnic 
Thank you for your fast answer.
I want some other information about Intel D415 camera. I want to use then for gesture and form recognition. My idea is to allow the camera to detect some movement or form directly.

teriterance, 2019-07-22:
Hi

closed 2019-07-22 by teriterance