realsenseai/librealsense #4474: Issue / Help : D435i IMU Reading And Depth Data Problem

TanmayPathak, 2019-07-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Platform                                 | Intel NUC |
| Language                            |  Python                        | 

### Issue Description
According to python wrapper of Intel Realsense SDK , following method gives me the raw data of IMU .

import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.accel)
config.enable_stream(rs.stream.gyro)

And for Depth data ...
 
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

Now problem with the raw data of IMU is that , Whenever I add the above line of code in program ( regarding streaming of  depth frame ) ; data which I am getting from the IMU are not stable in nature and are incorrect . 
And while removing this line [ config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30) ] , data which I am getting are much stable and accurate .

I have attached python code file bellow .

There is no other way to use depth and IMU data together ?  

[D435i.txt](https://github.com/IntelRealSense/librealsense/files/3416754/D435i.txt)

dorodnic, 2019-07-22:
Pipeline wait for frames will work at the lowest common rate (in this case 30 FPS). 
To get IMU data at full rate while also streaming video, please take a look at rs-callback example that shows how to process data asynchronously.

TanmayPathak, 2019-07-26:
Is there any method available in python for "process data asynchronously" ?

giovannicairoli, 2019-08-02:
Hello,

I have the same exact problem. In C++. Once I cancel the line config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30) everything works perfect. Otherwise sometimes the values are wrong(super big). 
Yes of course, you already suggest to use the Callback sample. But I'm a beginner and I don't know which are the APIs  that I need to you now to extract the depth, gyro and accel values. Is there any example?

Thank you very much for you patient,

Regards,

Giovanni

closed 2019-08-20 by TanmayPathak

stino78, 2020-09-08:
It is a little grose, but here's some code:

# -*- coding: utf-8 -*-
"""
Created on Tue Sep  8 22:25:35 2020

@author: EGO4
"""

import pyrealsense2 as rs
import numpy as np
import time
import cv2

def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

def initialize_camera():
    p_depth = rs.pipeline()
    p_imu = rs.pipeline()
    conf_depth = rs.config()
    conf_imu = rs.config()
    CAM_WIDTH, CAM_HEIGHT, CAM_FPS = 848,480,6
    conf_depth.enable_stream(rs.stream.depth, CAM_WIDTH, CAM_HEIGHT, rs.format.z16, CAM_FPS)
    conf_depth.enable_stream(rs.stream.color, CAM_WIDTH, CAM_HEIGHT, rs.format.rgb8, CAM_FPS)
#    conf_imu.enable_stream(rs.stream.infrared, CAM_WIDTH, CAM_HEIGHT, rs.format.y8, CAM_FPS)
    conf_imu.enable_stream(rs.stream.accel,rs.format.motion_xyz32f,250)
    conf_imu.enable_stream(rs.stream.gyro,rs.format.motion_xyz32f,200)
    prof_depth = p_depth.start(conf_depth)
    prof_imu = p_imu.start(conf_imu)
    return p_depth, p_imu

t0 = time.time()
pipe_depth, pipe_imu = initialize_camera()
print(time.time() - t0) # "seconds elapsed"

try:
    n = 0;
    i_num = 0
    d_num = 0
    while n<6000:
        frames_depth = pipe_depth.poll_for_frames()
        if frames_depth:
            depth_frame = frames_depth.get_depth_frame()
            color_frame = frames_depth.get_color_frame()

            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())
    
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
    
            images = np.hstack((color_image, depth_colormap))
    
            cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
            cv2.imshow('RealSense', images)
            cv2.waitKey(1)
            print("Depth ", d_num)
            d_num = d_num + 1
            
        frames_imu = pipe_imu.poll_for_frames()
        if frames_imu:
            accel = accel_data(frames_imu[0].as_motion_frame().get_motion_data())
            gyro = gyro_data(frames_imu[1].as_motion_frame().get_motion_data())
            print(i_num, " - accelerometer: ", accel, "\n",n, " - gyroscope: ", gyro,)
#            print("IMU ", i_num)
            i_num = i_num + 1
        
            n = n + 1
        print(n)
        
finally:
    pipe_imu.stop()
    pipe_depth.stop()
    cv2.destroyWindow('RealSense')

giovannicairoli, 2020-09-09:
Hello @stino78 , I will try it as soon as possible. Thank you