realsenseai/librealsense #4475: Crash when T265 is disconnected

seratg, 2019-07-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.15.0-20                                  | 
| Platform                                 | PC |
| SDK Version                            | 2.24.0 | 
| Language                            | C++ | 

### Issue Description
Hi, we are facing an issue with T265. When the device is being disconnected, the wait_for_frames() throws an exception; however, when we catch the exception and quit the application (which destroys our class that uses RealSense SDK), the application crashes.

The stack trace is shown as follows:
`#4 0x7f2d8f13043c librealsense::tm2_sensor::stop()
#5 0x7f2d8f0a3f81 librealsense::util::config::multistream::stop()#6 0x7f2d8f09b778 librealsense::pipeline::pipeline::unsafe_stop()
#7 0x7f2d8f09a8e4 librealsense::pipeline::pipeline::~pipeline()
#8 0x7f2d8f2541d7 __gnu_cxx::new_allocator<>::destroy<>()#9 0x7f2d8f253f23 std::allocator_traits<>::destroy<>()
#10 0x7f2d8f25373f std::_Sp_counted_ptr_inplace<>::_M_dispose()
#11 0x55f697df3776 std::_Sp_counted_base<>::_M_release()#12 0x7f2d8ed0ceb5 std::__shared_count<>::~__shared_count()
#13 0x7f2d8f0a4584 std::__shared_ptr<>::~__shared_ptr()
#14 0x7f2d8f0a45a0 std::shared_ptr<>::~shared_ptr()
#15 0x7f2d8f23b43c rs2_pipeline::~rs2_pipeline()
#16 0x7f2d8f21c55e rs2_delete_pipeline`

For T265, we enable RS2_FORMAT_6DOF stream.

We are also using the D415 camera and we have no issues with it. When the camera is disconnected, it throws an exception and then it exits properly.

I have also tried to exit the application by registering to:
`real_sense_context.set_devices_changed_callback([&](rs2::event_information& info) {
});`
but that also didn't solve the issue.

Thanks in advance

radfordi, 2019-07-23:
Thanks for your report.  `rs-pose` segfaulting on disconnect is definitely a bug.

ev-mp, 2019-09-26:
@seratg , do you still observe this behavior with the recent versions of the SDK ? 
Please update

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

@seratg , do you still observe this behavior with the recent versions of the SDK ?
Please update

seratg, 2019-10-22:
Hi, unfortunately I don't have access to the cameras anymore, therefore I won't be able to test/reproduce it

closed 2020-09-11 by RealSenseSupport