realsenseai/librealsense #4476: Question about passing T265 data frame content as a matrix to remap() in C++

HadeelRS, 2019-07-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC
| SDK Version                            |  2.23.0                         | 
| Language                            |  C++                          | 
| Segment			|  code on PC                          | 

### Issue Description
Hello dear,

I am trying to apply the code of T265_Stereo.py, in the following link, with C++ language:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

but I faced problem at the end of the code .. in the function remap () (at line 231) to un-distort and crop the center of frames. And the problem is:

The first parameter "src" which passed to the function remap(), it should be a matrix (e.g. cv::Mat) that have the values of fisheye frame data. But in fact the rs2::frame.get_data() return a pointer with type (const *void). 

And I don't know (1) how can I get the values or the content of the frame as a matrix rather than having the reference (I am not good in dealing with pointers and references). 
Or 
(2) how can I pass the pointer (const *void) as cv::Mat to the remap() and allow the function access the content/values directly.

So, could you help me in answering my question (1) or (2), please.

FYI:
 Here the remap() syntax:
void remap(InputArray src, OutputArray dst, InputArray map1, InputArray map2, int interpolation, int borderMode=BORDER_CONSTANT, const Scalar& borderValue=Scalar())
https://docs.opencv.org/2.4/modules/imgproc/doc/geometric_transformations.html?highlight=remap

Thank you in advance 
Hadeel
x

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

@HadeelRS - Have you been able to figure out answers to your questions on this ticket?

HadeelRS, 2019-08-29:
Yes I found a solution ... Thank you

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

Excellent

closed 2019-09-06 by RealSenseCustomerSupport