realsenseai/librealsense #4479: Depth detection with Unity wrapper.

kingbaggot, 2019-07-22:
[image]
| Camera Model | { D415 } |
| Firmware Version | (05.09.02.00) |
| Operating System & Version | {Win (10) / |
| Platform | PC |
| SDK Version | { 2.0 } |
| Language | {unity c# } |
| Segment	| {?? } |

### Issue Description
I want to detect where users touch on a projected on table. In my test I have two tripods - one at 2.3 m ( representing the head/shoulders of a user) & another at 2.9 m from the camera (which would be where their hands would be) . The projection surface is 3m away. 

Unity RealSense creates a depth texture than I put through opencv's blob detector and the result -  it easily detects the object that is nearer the camera (to the left of middle), but not the one nearer the projection surface (up and to the right from middle)

**Will I be able to somehow filter out these closer objects and better detect the objects nearer the projection surface (which are currently barely visible against the 3m surface) using the Unity SDK.**  If not do I have any other options ?

I've tried with the Nuitrack middleware, but because the camera is fixed vertically above the table it doesn't make out the skeletons.

I've also considered trying to track down a 200 series camera as I've read the 2016 unity RS sdk allowed access to more depth info.

any help with this would be awesome - have been getting a lot of help from Marty on the intel forum, but this feels very SDK specific.

MartyG-RealSense, 2019-07-22:
Thanks for your kind mention of me.  I just wanted to comment solely on the aspect of finding a camera compatible with the 2016 SDK.  You will have difficulty finding an official R200 or SR300 now.  The Creative BlasterX Senz3D camera is near identical to the SR300 and works with the 2016 SDK though, as Creative manufactured the official SR300 for Intel.

https://uk.creative.com/p/web-cameras/blasterx-senz3d

The Razer Stargazer is another SR300-compatible that works with the 2016 SDK.

ogoshen, 2019-11-14:
You could try calculating surface normals or doing a plane fitting to find the table top, finding the user's hands as points above the surface... either way you probably won't be able to do this with just image processing directly on the depth map.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@kingbaggot Any other questions with the reply from ogoshen? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@kingbaggot If no other questions we will close this. Thanks!

closed 2019-12-13 by RealSenseCustomerSupport