realsenseai/librealsense #4480: D435i: Cannot stream RGB at 60fps

pzoltowski, 2019-07-22:
| | |
|---|---|
|**librealsense**|2.24.0 RELEASE|
|**OS**|Mac OS|
|**Name**|Intel RealSense D435I|
|**Serial Number**|843112070584|
|**Firmware Version**|05.11.06.250|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|852323051288|

I cannot stream RGB data at 60 fps. I tried different resolutions, formats and even turned off all other streams (depth, ir). I have tested all those combinations with _realsense-viewer_ and it always shows at the top status bar (at the top of sensor image) maximum fps 30 even when chosen 60 fps. Changing to lower fps e.g. 15 fps works. Streaming at 60 or 90 fps also works for IR and Depth streams. I have only problem with color stream at higher frame rate than 30. I have tested fps with modified sample from opencv_viewer_example.py as well.

I use original usb cable and I have unplugged all other usb devices. Realsense-viewer detect camera as connected to usb 3.2 port. I'm using macbook pro 15 unibody 2012 that has usb 3.0

dorodnic, 2019-07-23:
Hi @pzoltowski 
Under RGB controls you have the Exposure Priority Mode. Try disabling it and see if it makes a difference. By default, the sensor is allowed to reduce FPS when it needs to increase exposure due to low lighting conditions.

pzoltowski, 2019-07-23:
Thx @dorodnic . I have tested again today in the morning during the day - some observations:
1) I got stable 60 fps when disabled _'enable auto exposure'_ altogether once. When I tried to change _'auto exposure priority'_ I get an error: 
```
UNKNOWN in rs2_set_option(options:0x7faab110e230, option:Auto Exposure Priority, value:6):
libusb_control_transfer(...) returned LIBUSB_ERROR_PIPE
```

2) When tested again in pretty much the same environment 10 minutes later. I keep getting 30 fps even when disabled auto exposure (tested both 640x480@60 and 320x240@60) - again tested with realsense-viewer and even unplugged realsense to cool down just to be double sure it's not any thermal throttling .

3) One time with good lighting condition I manged to get 60fps even with default auto exposure enabled in the beginning but only for a while. While camera still in same place and without any ambient light changing it dropped to 30 fps when e.g. waved once my hand and then stayed at 30fps - hard to reproduce it and and get any robust 60fps with RGB stream

4) Is there any hope that e.g. 320x240@90 can be introduced in future firmware or is there any hardware limitation? I need high fps and low latency for precise motion tracking and would prefer to use realsense rather than playstation 3 eye camera.

ev-mp, 2019-09-26:
@pzoltowski , a note regarding 90 fps:
At least with the [FW5.11.11.100](https://dev.intelrealsense.com/docs/firmware-updates) the following  60+ fps modes are available:
   Depth         848x480       @ 90Hz     Z16
   Depth         848x100       @ 100Hz    Z16
   Depth         640x480       @ 90Hz     Z16
   Depth         640x360       @ 90Hz     Z16
   Depth         480x270       @ 90Hz     Z16
   Depth         424x240       @ 90Hz     Z16

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

This issue can be observed, will further check internally.

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

Hi @pzoltowski

Can you test with the latest librealsense 2.31.0 release and latest FW 5.12.1 release?  This issue is no longer observed with the latest updates.

pzoltowski, 2020-01-03:
Thanks, I will test it this weekend and give a feedback

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @pzoltowski

Were you able to test this out?

RealSenseCustomerSupport, 2020-01-13:
----------------------------------------------

Hi @pzoltowski

Let us know if you're able to test this out.  If no response then we will go ahead and close out the thread.  If anything further is necessary with the latest SDK and FW releases, then you can go ahead an open a new thread.

closed 2020-01-24 by RealSenseCustomerSupport