realsenseai/librealsense #4482: pointcloud in camera coordinate and world coodiante

yutaroyamanaka, 2019-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PC |
| Language                            |  python                         | 

### Issue Description
in librealsense/wrappers/python/examples/opencv_pointcloud_viewer.py
I can get the verts.

I cat get the verts by using following code.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L297

```
# Pointcloud data to arrays
v, t = points.get_vertices(), points.get_texture_coordinates()
verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz
```
 
I know I can get the data from gyroscope and accelerometer to compute the rotation by using D435i IMO, however, I am not sure how to convert xyz in the code to that in the world coodinate.

yutaroyamanaka, 2019-07-23:
Is following equation  available in this case ? 
(α, β, γ) is the sensor attitude angle and (Xw, Yw, Zw) is the point in world coodinate
[image]

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@yutaroyamanaka Sorry for late response.  Please check https://github.com/IntelRealSense/librealsense/issues/5063 to see if it helps your question. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@yutaroyamanaka Any other questions? Looking forward to your update. Thanks!

closed 2019-12-14 by yutaroyamanaka