realsenseai/librealsense #4483: AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

swjtuccy, 2019-07-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435}| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  jetson nano| 
| Kernel Version (Linux Only)                     | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            | sdk2.0                          | 
| Language                            |  python3}                          | 

### Issue Description
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

I have build librealsense and used pip3 install pyrealsense2-aarch64, everything is right .But when i try to run the example python scripts , the Error ocured as:
:~/.local/bin$ python3 align-depth2color.py 
Traceback (most recent call last):
  File "align-depth2color.py", line 16, in <module>
    pipeline = rs.pipeline()
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

lramati, 2019-08-21:
I'm not sure what the pyrealsense2-aarch64 package is or who made it, but downloading the wheel from PyPi and opening it with 7z reveals that it contains only some example scripts, and not the actual library.
Try uninstalling pyrealsense2-aarch64 and install pyrealsense2 instead.

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@swjtuccy Did you try what Iramati suggested? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@swjtuccy Any update about this? Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@swjtuccy If no other questions, will close it at this point. Thanks!

closed 2019-10-12 by RealSenseCustomerSupport

darmin-hsiung, 2020-11-06:
I have install pyrealsense, but run some program was error, the question as follow, please help me to solve this question, thanks!
$ python demo.py

Traceback (most recent call last):
  File "demo.py", line 1, in <module>
    import pyrealsense2 as rs
  File "/home/u16l/WORKS/Codes/pyrealsense2.py", line 6, in <module>
    pipeline = rs.pipeline()
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'