realsenseai/librealsense #4484: Intel Realsense Driver for Debian 9

millto, 2019-07-23:
Hello togehter, 
I would like to run a Intel RealSense D435 camera on a Linux 4.9.0-9-rt-amd64. 
Is there a driver for this? If not, is there a way to build a driver on yourself?
I tried to to port a 4.4 or 4,10 kernel patch to Debian 9, but it did not work..
Can you help me?

ev-mp, 2019-07-28:
@millto hi, we don't have a dedicated kernel patch for DebianStretch. Instead we provide reference kernel patches for most major Ubuntu LTS builds that can be ported by community.

Porting the patches for a different distro is an engineering task, so If you'd  share where you stand with the port and describe the issues you encounter then someone on the forum will probably assist you.

Note that the kernel patch is not a prerequisite for using the device with any distro using a modern (4.4+) kernel, as all the basic controls and streaming modes are fully functional. The added values for applying the patch are:
- UVC and HID (IMU) HW timestamps
- Frame metadata attributes ( gain/exposure, projector power, white balance, etc')
- Minor control fixes.
- Minor kernel fixes (may require certain FW version for compatibility)

Also note that most of those functions can be obtained by using the alternative `lubuvc` backend, which is a user-space implementation of the UVC driver layer.

RealSenseCustomerSupport, 2019-08-21:
----------------------------------------------

@millto With the clarification from ev-mp, do you have any other questions for this ticket? Thanks!

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

@millto Could you please update? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-09-10:
----------------------------------------------

@millto If no other questions will close this accordingly. Thanks!

closed 2019-09-10 by RealSenseCustomerSupport