realsenseai/librealsense #4486: Intel RealSense D400 series documentation changes

AntoineWefit, 2019-07-23:
### There is some differences between versions of the D400 series documentation
I noticed that there is an inconsistency between the values of the **Infrared Projector's FOP**, both wide and standard, in the 2 different versions of the documentation attached (March 2018 & June 2019).

Values can be found pages 34-35 (2018) & 35-36 (2019).

Was the IR emitter changed ? I need to know the exact values for my application.

[Intel-RealSense-D400-Series-Datasheet-Jun-2019.pdf](https://github.com/IntelRealSense/librealsense/files/3421315/Intel-RealSense-D400-Series-Datasheet-Jun-2019.pdf)
[Intel-RealSense-D400-Series-Datasheet-Mar-2018.pdf](https://github.com/IntelRealSense/librealsense/files/3421316/Intel-RealSense-D400-Series-Datasheet-Mar-2018.pdf)

NXTwoThou, 2019-08-05:
Just as someone who's interested to know the answer too, thought I'd put the numbers directly in the thread:

2018 Standard
Horizontal Field of Projection 80
Vertical Field of Projection  55
Diagonal Field of Projection  89.3
2019 Standard
64+/- 3
41+/-3
72+/-3
Losing a -minimum- of 14.3 degrees diagonal feels pretty massive.

2018 Wide
100.4
69
110.4
2019 Wide
86+/-3
57+/-3
94+/-3
On the wide, losing a minimum of 13.4 degrees.

Could this be a D415 versus D435 difference?  I could see the D415 having a narrower FOP

AntoineWefit, 2019-08-05:
> Could this be a D415 versus D435 difference? I could see the D415 having a narrower FOP

The D415 and D435 are indeed not using the same projector, but the problem remains :)

[image]

NXTwoThou, 2019-08-05:
Derp on me, need moar coffee. :D

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

Hi @AntoineWefit

The data within the latest datasheet is the most up to date information regarding the IR projector FOV

Thanks

closed 2019-09-18 by RealSenseCustomerSupport