realsenseai/librealsense #4487: query_devices() BUG

eMrazSVK, 2019-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.4.38 | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            | 2 | 
| Language                            |  C++                         | 
| Segment			|  Robot | 

### Issue Description
Hello, 

I am writing ROS wrapper for T265 and D435 only, but it doesn't really matter I guess.

I found out, that method `query_devices()` does not always work as it is supposed to. Whenever when I initialize any `rs2::` object, lets say `rs2::pipeline` for example, `query_devices()` method is not able to return T265 (it returns list with size of 0). However, when I call `query_devices()` first (before initializing any `rs2::` object or anything), it returns descriptor of T265. When I plug in lets say D435 and D415 at the same time it works fine even with initializing `rs2::pipeline` before calling `query_devices()`. 

It is very strange behaviour. Can someone reproduce this, or confirm the bug? I am no such expert to debug this problem.

Thank you for assistance.

ev-mp, 2019-07-28:
@eMrazSVK, there is a certain limitation for the supported T265 operation modes-
In multi-cam scenarios (including one or more T265), all the T265 devices will be mapped, and consequently made available for querying via the very first `rs2::context` object created in code.

This a known issue elaborated in #4163, #4261.

There is [work in progress](https://github.com/IntelRealSense/librealsense/pull/4302)  by a community member @orshefi to alleviate this constrain. You're welcomed to review and provide contributions as well.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi @eMrazSVK

Is there anything further that is needed here that isn't covered in the response from ev-mp?

Thanks

closed 2019-09-13 by RealSenseCustomerSupport