realsenseai/librealsense #4489: RealSense, OpenCV & Raspberry-Pi

pkr97, 2019-07-23:
Hello @dorodonic , I want to use Intel RealSense D415, OpenCv, but inside Raspberry-Pi and preferrably c++ because of speed as a concern. 

I found this link https://dev.intelrealsense.com/docs/using-depth-camera-with-raspberry-pi-3.
It mentions that I can get 30FPS.

Is it possible to get 60FPS with 320x240 sized image? 

Thanks..

MartyG-RealSense, 2019-07-23:
The table below lists that modes that are supported for USB2.  According to the tables below, 480x270 for depth and 424x240 for color are the lowest available resolutions.  

[image]

[image]

ev-mp, 2019-08-05:
@pkr97 , you can also run `rs-enumerate-devices -m` to list all the D415-supported streaming profiles.

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@pkr97 Any other questions about this? Looking forward to your reply. Thanks1

closed 2019-09-29 by RealSenseCustomerSupport