realsenseai/librealsense #4490: How to reliably set the auto-exposure

AntoineWefit, 2019-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 2.24 | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  /                                     | 
| Platform                                 | PC|
| SDK Version                            |  latest                         | 
| Language                            |  C++                         | 
| Segment			|  others                      | 

### I would like to know how to correctly set the auto-exposure
My application require the use of the auto-exposure via a C++ file.
My problem is that the data are captured too quickly and thus the auto-exposure doesn't have the chance to settle :(

We currently use this line of code to wait a bit : 
`for (auto i = 0; i < 30; ++i) pipe.wait_for_frames();`
But it's not really reliable and I would like to know if there is another/better way to do that.

dorodnic, 2019-07-23:
Information about actual value of exposure is provided within frame metadata. To ensure it is available, you need to run a powershell script from scripts directory or install the official UWP driver.
Once you have metadata, you can monitor actual exposure value and ensure it is stable.

AntoineWefit, 2019-07-24:
Thank you for the reply @dorodnic ! I'll take a look at that :)

To be a little bit more specific, we are using the camera in an "independent" system with no display of any kind.
We're capturing data (images & 3D) with the execution of a custom code, with just a click from the user; so almost automatically. The camera is tuned, capture start, data saved, and then sent to the user.
What I would like to do is to find a way to allow the auto-exposure to settle reliably and without the intervention of a person.
**Would you consider that waiting for more frames is the correct way to do so ?** I don't really want to extend the execution time too much...

In any cases, I'll try what you suggested.

ev-mp, 2019-08-05:
@AntoineWefit , once you have the metadata setup then you can monitor the actual exposure data till the point that it converges (i.e. the exposure changes in the last N frames is below some TBD threshold). This flow will automatically adjust for the actual light condition and the stream fps.

Note that the actual exposure attribute is available for Depth and IR streams only.

AntoineWefit, 2019-08-06:
> @AntoineWefit , once you have the metadata setup then you can monitor the actual exposure data till the point that **it converges (i.e. the exposure changes in the last N frames is below some TBD threshold)**. This flow will automatically adjust for the actual light condition and the stream fps.

Ah yes ! This is what I'm looking for :)

> Note that the actual exposure attribute is available for Depth and IR streams only.

Does this mean I can't do the same thing for the RGB stream ? I'll have to find a workaround then, because the auto-exposure for color/texture works well on daylight conditions, but not so great when there is shadows. Plus I can't reliably set a fixed value as the picture will sometimes be overexposed, or worse, trigger a USB overflow...

ev-mp, 2019-08-06:
Unfortunately you're correct - the D400 RGB sensor does not emit the actual exposure value when AE mode is activated. #2549

AntoineWefit, 2019-08-06:
Alright, thank you for your answer and the link ! I'll try to write some code, at least for the Depth & IR streams, to monitor the exposure. I'll let this issue open a few days in case I can't get it to work.

Might try [that](https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387), I'll keep you updated.

AntoineWefit, 2019-08-08:
Turns out I mostly need to control the auto-exposure for the texture (so the RGB sensor)...
Didn't find anything yet to do so, but I'm closing this issue as my question has been answered.

closed 2019-08-08 by AntoineWefit