realsenseai/librealsense #4491: Realsense Driver didn't work with NiTE2

choonyip, 2019-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-55 | 
| Platform                                 | PC |
| SDK Version                            |  v2.24.0  | 

### Issue Description
Realsense Driver (```librs2driver.so```) worked with OpenNI2, but didn't work with NiTE2. Tried Realsense Driver with ```./OpenNI-Linux-x64-2.3/Samples/Bin/SimpleViewer```, it worked correctly. Then, I tried with ```./NiTE-Linux-x64-2.2/Samples/Bin/SimpleUserTracker```, but it didn't work. It output error ```Could not get shift2Depth table.```

### Output of ./SimpleUserTracker
```cy8182@cy8182-TM1604:~/backup/NiTE-Linux-x64-2.2/Samples/Bin$ ./SimpleUserTracker 
[RS2] +Rs2Driver
[RS2] initialize 
[RS2] enumerateDevices 
[RS2] devicesChanged removed=(nil) added=0x12b17a0
[RS2] deviceConnected
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=843112071902 name=Intel RealSense D435I
[RS2] Rs2Driver INITIALIZED
[RS2] deviceOpen uri=843112071902
[RS2] +Rs2Device
[RS2] initialize 
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=843112071902 name=Intel RealSense D435I
[RS2] initializeStreams 
[RS2] SENSOR 0
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=1280 height=800 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=1280 height=800 framerate=30
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=10 width=1280 height=800 framerate=25
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=10 width=1280 height=800 framerate=25
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=10 width=1280 height=800 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=10 width=1280 height=800 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=1280 height=800 framerate=15
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=1280 height=800 framerate=15
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=1280 height=720 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=1280 height=720 framerate=30
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=1280 height=720 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=1280 height=720 framerate=15
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=1280 height=720 framerate=6
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=1280 height=720 framerate=6
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=848 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=848 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=848 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=848 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=848 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=848 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=848 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=848 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=848 height=480 framerate=6
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=848 height=480 framerate=6
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=848 height=100 framerate=100
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=848 height=100 framerate=100
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=640 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=640 height=480 framerate=90
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=640 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=640 height=480 framerate=60
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=640 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=640 height=480 framerate=30
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=640 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=640 height=480 framerate=15
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=9 width=640 height=480 framerate=6
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=9 width=640 height=480 framerate=6
[RS2] 	type=3 sensorId=0 streamId=2 profileId=9 format=10 width=640 height=400 framerate=25
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=10 width=640 height=400 framerate=25
[RS2] 	type=3 sensorId=0 streamId=1 profileId=8 format=10 width=640 height=400 framerate=15
[... 361 lines cut ...]
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=4 width=320 height=240 framerate=30
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=5 width=320 height=240 framerate=6
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=10 width=320 height=240 framerate=6
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=4 width=320 height=240 framerate=6
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=5 width=320 height=180 framerate=60
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=10 width=320 height=180 framerate=60
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=4 width=320 height=180 framerate=60
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=5 width=320 height=180 framerate=30
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=10 width=320 height=180 framerate=30
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=4 width=320 height=180 framerate=30
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=5 width=320 height=180 framerate=6
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=10 width=320 height=180 framerate=6
[RS2] 	type=2 sensorId=1 streamId=0 profileId=10 format=4 width=320 height=180 framerate=6
[RS2] mainLoop 
[RS2] getSensorInfoList 
[RS2] createStream sensorType=3
[RS2] start type=1 sensorId=0 streamId=0
[RS2] updateConfiguration 
Couldn't get shift2Depth table.
Segmentation fault (core dumped)
```

Already recompiled driver with patch
<https://gist.github.com/wongfei/104e5b382f82295deed0f38a07e9a7f3>
as described in issue #2825, but still getting the same output error.

choonyip, 2019-07-26:
[Rs2Device.cpp](https://github.com/choonyip/librealsense/blob/master/wrappers/openni2/src/Rs2Device.cpp)
[Viewer.cpp](https://github.com/choonyip/NiTE-2.0.0/blob/master/Samples/UserViewer/Viewer.cpp)
[Viewer.h](https://github.com/choonyip/NiTE-2.0.0/blob/master/Samples/UserViewer/Viewer.h)
Changed the above files as stated in #2825, but getting the same output error.

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@choonyip Sorry for late reply. Did you still have this issue with the latest SDK?

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@choonyip Could you please update? Thanks!

choonyip, 2019-12-03:
Thanks for the reply, but sorry, I'm no longer using this sensor.

closed 2019-12-12 by RealSenseCustomerSupport