realsenseai/librealsense #4495: Context instantiation throws exception

hopaa, 2019-07-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Win 10 (10.0.17134)  | 
| Platform                                 | PC  |
| SDK Version                            |  2.24.0                           | 
| Language                            |  C#                          | 
| IDE                                      | Visual Studio 2015 |

Hi everybody,

I built and ran .net examples successfully. But when I tried to create my own .net project, something went wrong. I can't figure it out what should I do to resolve it!

I just copied and pasted the "cs-tutorial-4-software-dev" source files somewhere else and added them and the required reference assemblies to my WPF project. Whereas it was built successfully, when I tried to run it, this exception occurred:
"The type initializer for 'Intel.Realsense.Context' threw an exception."

I traced the code, when it tries to get an instance of a Pipeline, it calls a line of code that wants to get an instance of a Context which throws the exception.

I mean this line of code throws the exception (line 35 of Pipeline.cs):
`using (var ctx = new Context())`

Any idea how to handle it?!

Thanks in advance

dorodnic, 2019-07-25:
There are couple of possible reasons for such crash - 
1. Make sure the .NET RealSense assembly and the C++ realsense2.dll are in the same directory, or can be otherwise found (by adjusting PATH env variable). Also make sure there is no mismatch between versions of these two DLLs. 
2. When building your .NET project with native dependencies, you cannot leave your project "Any CPU", as it gives .NET the freedom to pick either x86 or x64 architecture at runtime. If you are using x64 version of the realsense2.dll your App must specify x64 in its settings, and same for x86.

hopaa, 2019-07-25:
Thank you @dorodnic. Actually the second one resolved my problem. I really appreciate that.

closed 2019-07-25 by hopaa