realsenseai/librealsense #4496: rs2::pipeline.start(rs2::config) timeout

galikrosentalik, 2019-07-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |  Ubuntu 18| 
| Kernel Version (Linux Only)    |         4.18.0-25-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.24.0 | 
| Language                            |  c++| 
| Segment			|  others | 

### Issue Description
Hi everybody,
I am working with the SDK. My system working flow requires the user to load a config file that
specifies the camera serial number.
In case of a wrong serial number in the config file, I catch the exception thrown from the "rs2::pipeline.start(rs2::config)" method.
At the moment, the exception is thrown 15 seconds after calling the method. I would like to control the amount of time it takes for the exception to be thrown (e.g. 1 second instead of 15), and I cannot find the way to do this. 
As an example the "rs2::pipeline.wait_for_frames" method can accept a timeout value as a parameter. I would like to acheive similar result with the "rs2::pipeline.start(rs2::config)" method.
Can anyone assist please?

Thanks, Gal.

dorodnic, 2019-07-25:
At the moment this value is not configurable, but we can add such API in a future release. And if you must control the timeout, I can point you to the places in code it appears and you could build custom version of the SDK from source.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi @galikrosentalik

dorodnic provided some comment on this matter.  At this point in time is there anything further needed?

Thanks

abedMeto, 2020-08-25:
Hi,
I work with pylibrealsense. regarding pipline.start(rs2::config) timeout
Is it still not configurable?

Thanks,
Abed

RealSenseSupport, 2021-03-19:
Thank you for highlighting this enhancement. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265. This will not be configurable.

closed 2021-03-19 by RealSenseSupport