realsenseai/librealsense #4497: How to simultaneuosly stream the frame from the intel camera and send serial commands??

Ahad19931, 2019-07-24:
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC  | 
| Language                            |  {matlab }                          | 
| Segment			|  {Robot}                          | 

I am having a problem of continuously sending serial commands while performing other imgae processing steps on a certain frame accquired from a camera. My task is to continuosly get the frames from the camera and then using multiple image processing techniques, I need to simplify the frame and then using the area of the object present in that binary frame, I need to navigate the robot to perform obstacle avoidance task. For this purpose the serial communication between the robot and MATLAB needs to be done continuously and when a certain object, lets say an object with a area of 40000(white pixels) is detected in the frame, the robot stops. Below you can find the code which performs the above task.
function align_cust_example()
% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Start streaming on an arbitrary camera with default settings
profile = pipe.start();
align_to = realsense.stream.color;
alignedFs = realsense.align(align_to);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
while(true)
for i = 1:5
fs = pipe.wait_for_frames();
end
% Select depth frame
aligned_frames = alignedFs.process(fs);
depth = aligned_frames.get_depth_frame();
% get depth image parameters
depthSensor = dev.first('depth_sensor');
depthScale = depthSensor.get_depth_scale();
depthWidth = depth.get_width();
depthHeight = depth.get_height();
% retrieve UINT16 depth vector
depthVector = depth.get_data();
% reshape vector, and scale to depth in meters
depthMap = double(transpose(reshape(depthVector, [depthWidth,depthHeight]))) .* depthScale;
[r,c]=size(depthMap);
      for i=1:r
        for j=1:c
            if(depthMap(i,j)>=0.2 && depthMap(i,j)<=0.45)
                depthMap(i,j)=1;
        else
                depthMap(i,j)=0;
            end
        end
      end
%align the depth frames to the color stream
%Select rgb frame
color = fs.get_color_frame();
colordata=color.get_data();
img = permute(reshape(colordata',[3,color.get_width(),color.get_height()]),[3 2 1]);
%simplify binary image using morphological operations
img_fill=imfill(depthMap,'holes');
s=strel('disk',25);
imgM=imclose(img_fill,s);
img_area=50;
Iout = bwareaopen(imgM,5000);
imshow(Iout)
cc=Iout;
my_area=struct2cell(regionprops(cc,'Area'));
area=cell2mat(my_area)
%Start serial communication with robot
while(true)
delete(instrfind);
s=serial('COM9','Baudrate',57600,'Databits',8,'StopBits',1,'parity','none');
fopen(s);
if(isempty(area))
    %pause(100/1000);
    val=hex2dec({'AA','0A','00','00','55'});
    fwrite(s,val);
else
    [k,l]=size(area);
    for row=1:k
       for col=1:l
          if(area(row,col)<=40000)   
            %pause(100/1000);
            val=hex2dec({'AA','0A','00','00','55'});
            fwrite(s,val);
          else
            %pause(100/1000);
            val=hex2dec({'AA','00','00','00','55'});
          end
       end
    end
end
fclose(s);
end
end
% Stop streaming
pipe.stop();
end
However, the problem is that, with this code, there is a noticible delay because the frame is first taken from the camera, it is processed and then serial communication starts which send the required serial command (e.g. 'AA' '00' '00' '00' '55') according to the satisfied condition and then robot takes a ceratin iteration and stops. Then again, the process is repeated and another command is send to the robot for it to move further. Due to this the robot does not continously moves which is the actual task.
Therefore, I need help to make this code efficient enough so that the robot continuously gets the command while performing the frame simplication task simultaneously. I hope that this explanation gives a clear understanding for you to address the problem.
Note: ('AA' '00' '00' '00' '55') are the hexa values which needs to be send together to let the robot make a single movement.

MartyG-RealSense, 2019-07-24:
I wonder if the check needs to be done every frame.  If, for example, you are running the robot at around 5 frames per second then I would think that even if the robot's 'should I stop' status is checked once per second instead of 5 per second, that should be a sufficient reaction time to stop before collision, especially given that the camera should be able to detect the obstacle before the robot physically reaches it.  

So if the robot is using a 'for i = 0 to 5' type For loop, then the check could be put outside of the loop so that it initiates when frame 5 is reached, before the For loop starts the 5-long frame counting cycle again for the next second.

Robot motion may be smoother if its motion is continuously turned on until a stop condition is detected, instead of moving in micro-steps on each frame.  I encountered this issue when developing a videogame.  It seemed more realistic to make the player's character walk step by step like a real person does.  However, the result was very juddering.  It felt better to have the character move continuously for as long as the 'move' input was active and there was not an obstacle-detected condition currently true.

Ahad19931, 2019-07-25:
I tried minimizing the for loop iteration but still the robot runs the command once and stops for the next frame to give the area value to run the command again. The problem is that if I run the robot continuously, lets say, in a while(true) condition for e.g.
%Start serial communication with robot
while(true)
delete(instrfind);
s=serial('COM9','Baudrate',57600,'Databits',8,'StopBits',1,'parity','none');
fopen(s);
if(isempty(area))
    pause(100/1000);
    val=hex2dec({'AA','15','00','00','55'});
    fwrite(s,val);
else
    [k,l]=size(area);
    for row=1:k
       for col=1:l
          if(area(row,col)<=40000)   
            pause(100/1000);
            val=hex2dec({'AA','15','00','00','55'});
            fwrite(s,val);
          else
            pause(100/1000);
            val=hex2dec({'AA','00','00','00','55'});
          end
       end
    end
end
The variable'area' value will not be updates since no next frame is taken from the camera. What amendment should I do to run the robot continuously and stop as soon as the certain condition is met.

MartyG-RealSense, 2019-07-25:
The question of how to control the robot motion is somewhat outside of my field of knowledge, unfortunately.   I will link a couple of the RealSense robotics guys into this conversation to highlight it to them.

@doronhi @schmidtp1

doronhi, 2019-07-31:
I think this question is beyond the scope of this repository.
It is a programming question. A matlab programming question, if you insist on using matlab.
I agree with @MartyG-RealSense , you'd better have commands of "start moving" and "stop moving" to have a smooth moving robot.

If you can only use a "step" command for the robot, the first thing to do would be to change the structure of your code as follows:
```
% initialization:
for i = 1:5
fs = pipe.wait_for_frames();
end

while true:
    fs = pipe.wait_for_frames();
    % perform image processing task.
    % Send command to robot
```
No need to wait 5 frames every loop.

However, to have a smooth movement using a "step" command, the better idea would be to use 2 threads:
1. Thread 1: read frames and issue a start/stop command.
2. Thread 2: Continuously send "step" commands, until stopped by the first thread.

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

@Ahad19931 With the suggestions from MartyG-RealSense and Doronihi, do you have any other questions?  Thanks!

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

@Ahad19931 Any update for this? Looking forward to your update. Thanks!

Ahad19931, 2019-09-09:
The problem was solved. Thanks!

closed 2019-09-18 by RealSenseCustomerSupport