realsenseai/librealsense #4498: Accessing Raw Depth Images on Android through Java APIs

YunfanZhang42, 2019-07-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | macOS | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android |
| SDK Version                            |  2.24                         | 
| Language                            |  Java                         | 
| Segment			|  Smartphone                        | 

### Issue Description
Hi there, 

I am trying to use the Java API for Android to access the raw depth frames. I read that the RealSense camera will record the depth image in Z16 format. Since Android does not support Z16 as a native format for bitmaps, I am trying to convert the Z16 frames into ARGB 8888 format, saving the low 8 bits of each depth pixel to the green channel and high 8 bits into the red channel. However, it seems I can only access the low 8 bits for each of the pixels. Is there something I am missing here? Could the memory padding for the byte array be an issue? If so, what would be a good way to access 16 bit raw depth images?

Here is the code I use to convert Z16 to ARGB 8888:
~~~ java
// z16 comes from frame.getData()
public static int[] z16ToArgb(byte[] z16, int width, int height) {
        int frameSize = width * height;
        int[] argb = new int[frameSize];
        int index = 0;
        for (int j = 0; j < height; j++) {
            for (int i = 0; i < width; i++) {
                int A = 255;
                int B = 0;
                int G = z16[2 * index];
                int R = z16[2 * index + 1];
                argb[index] = (A & 0xff) << 24 | (B & 0xff) << 16 | (G & 0xff) << 8 | (R & 0xff);
                index++;
            }
        }
        return argb;
    }
~~~

And here is a sample image that I get. It seems the high 8 bits are missing, as there is no red channel on the image.

dorodnic, 2019-07-25:
Hi @YunfanZhang42 
I think you are doing everything right. Try multiplying the R value by some constant (say 16) and check if you see something (in this method red would represent far away depth)
Since green is encoding the low bits of depth it cycles between all possible values, while red have small number of possible values at longer distances.

YunfanZhang42, 2019-07-25:
Yep, it seems there is indeed non-zero values in the red channel, but the value is usually small so it's hard to tell from the pictures. Thank you!

closed 2019-07-25 by YunfanZhang42

vshalkov, 2019-12-13:
@YunfanZhang42 Please help me. How i can to get this image and save to file (.tiff) ?
[image]
I asked a question here, maybe you can help me, I will be grateful
https://github.com/IntelRealSense/librealsense/issues/5446

mei123hao, 2021-02-02:
@YunfanZhang42  Have you solved this problem ? I try to get the depth gray scale image. My code is following: 
`processed.getData(depth_bits);
depth_Buffer.clear();

int[] depth = new int[640 * 480];
for (int i = 0; i < depth.length; i++) {
    //关键代码，生产灰度图
    depth[i] =  depth_bits[2 * i] << 8 | depth_bits[2 * i + 1] | 0xFF000000;
}
depthbitmap = Bitmap.createBitmap(depth, 640, 480, Bitmap.Config.ARGB_8888);`
But it seems that the gray image is not correct.