realsenseai/librealsense #4499: Different between .bag file size recorded by D435 and D435i model

gachiemchiep, 2019-07-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  4.15.0.15                                         | 
| Platform                                 | PC |
| SDK Version                            |   2.24 }                          | 
| Language                            |  Cpp                          | 
| Segment			|  {others }                          | 

### Issue Description
I use the rs-record-playback Sample to record realsense stream into file. 
For a 5 seconds length data, D435 save about 156MB. 
But D435i save about 600MB.

My question is: 
1. What is the different between stream channels of D435 and D435i.

Using realsense-viewer, i can see that there's addition stream called motion-stream on D435i. I want D435 and D435i take the same amount of data. Is there any way i can turn this motion stream off programmingly on D435i.

I tried to turn off RS2_STREAM_FISHEYE, RS2_STREAM_GYRO, RS2_STREAM_ACCEL by 

```cpp
		if ("Intel RealSense D435I" == std::string(dev.get_info(RS2_CAMERA_INFO_NAME))) {
			std::cout << "Disable motion channel of D435i" << std::endl;
			cfg.disable_stream(RS2_STREAM_FISHEYE);
			cfg.disable_stream(RS2_STREAM_GYRO);
			cfg.disable_stream(RS2_STREAM_ACCEL);
		}
```

But no -luck. 

Here's all of the stream definition, did i miss something ?

```bash
/** \brief Streams are different types of data provided by RealSense devices. */
typedef enum rs2_stream
{
    RS2_STREAM_ANY,
    RS2_STREAM_DEPTH                            , /**< Native stream of depth data produced by RealSense device */
    RS2_STREAM_COLOR                            , /**< Native stream of color data captured by RealSense device */
    RS2_STREAM_INFRARED                         , /**< Native stream of infrared data captured by RealSense device */
    RS2_STREAM_FISHEYE                          , /**< Native stream of fish-eye (wide) data captured from the dedicate motion camera */
    RS2_STREAM_GYRO                             , /**< Native stream of gyroscope motion data produced by RealSense device */
    RS2_STREAM_ACCEL                            , /**< Native stream of accelerometer motion data produced by RealSense device */
    RS2_STREAM_GPIO                             , /**< Signals from external device connected through GPIO */
    RS2_STREAM_POSE                             , /**< 6 Degrees of Freedom pose data, calculated by RealSense device */
    RS2_STREAM_CONFIDENCE                       , /**< 4 bit per-pixel depth confidence level */
    RS2_STREAM_COUNT
} rs2_stream;
```

dorodnic, 2019-07-25:
If you are using the Viewer, you can enable compression under playback and record preferences in settings window. 
ROS-bag compression saves up on size a bit, but introduces some amount of frame drops. For IMU streams at high frequency this could cause degradation of tracking, so we disabled compression by default for the 35i.

gachiemchiep, 2019-07-25:
Hello @dorodnic 
Thank you for your information. 
So it mean that D435 turn the compression on by default and D435i turn the compression off by default.
That make sense. 
I have to save of lot of depth, color () into disk so i need this feature.

Sorry i don't find any oftion inside realsense-viewer that could help me turn the compression on....
 
Is there any way that i can turn the compression on for D435i by programming or using preset ?
I look through preset files at : 
[D400 Series Visual Presets](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) but couldn't find any related parameters.

dorodnic, 2019-07-25:
#3089 shows the relevant UI. 
Programmatically, it is possible, but a bit clunky. There is no easy way to control it via the pipeline. Let me know if you still need it, I'll try to guide you through this.

gachiemchiep, 2019-07-25:
Hello  @dorodnic . 
Thank you for your information. i found the button. And i'm able to turn the compression on using the realsense-viewer. 

I still need to turn on/off the compression programmingly, please help.

dorodnic, 2019-07-25:
The process would be:
1. Find the device you are using by creating a context and calling query_devices
2. Create recorder using [this ctor](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_record_playback.hpp#L228) by passing the output filename, device to be recorded, and compression flag.
2. Iterate through the sensors of the record device via query_sensors.
3. For each sensor, find the stream profiles you are interested in (say, Z16 640x480 @60 FPS)
4. Call sensor.open with profile / profiles you selected.
5. Create syncer object.
6. Call sensor.start on all sensors and pass syncer object as parameter to start.
7. Stream will be recorded. You can use syncer.wait_for_frames somewhat similar to pipeline.wait_for_frames, but please note that pipeline will always return frames from all streams, while the syncer can also provide "partial" framesets when full match could not be accomplished. 

You can take a look at sensor control example to understand this process better.

gachiemchiep, 2019-07-25:
hello @dorodnic 
Thank you for your information.

Actually, it's totally different with the approach i'm using right now. 
Is there any example that i can follow/

Currently, i'm following the approach used in [record-playback](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback) . 
Mainly using rs2::pipeline and rs2::config. 

```cpp
pipe_1->stop(); // Stop the pipeline with the default configuration

// Re Init the pipe: clear and release everything
pipe_1 = std::make_shared<rs2::pipeline>();

rs2::config cfg;
// re create the conf
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
auto advanced_mode_dev = dev.as<rs400::advanced_mode>();
// Check if advanced-mode is enabled to pass the custom config
if (!advanced_mode_dev.is_enabled()) {
	// If not, enable advanced-mode
	advanced_mode_dev.toggle_advanced_mode(true);
	std::cout << "Advanced mode enabled. " << std::endl;
} else {
	std::cout << "Advanced mode is already enabled. " << std::endl;
}

// Select the custom configuration file
std::cout << "Sensor will load configuration from" << preset << std::endl;
std::ifstream input(preset);
std::string preset_json((std::istreambuf_iterator<char>(input)), std::istreambuf_iterator<char>());
advanced_mode_dev.load_json(preset_json);
input.close();

// Reduce size of stream
std::cout << "Reduce the resolution of sensor using conf at : " << conf << std::endl;
boost::property_tree::ptree pt;
boost::property_tree::ini_parser::read_ini(conf, pt);
cfg.enable_stream(RS2_STREAM_COLOR, std::atoi(pt.get<std::string>("Color.width").c_str()),
				std::atoi(pt.get<std::string>("Color.height").c_str()), RS2_FORMAT_BGR8,
				std::atoi(pt.get<std::string>("Color.fps").c_str()));
cfg.enable_stream(RS2_STREAM_DEPTH, std::atoi(pt.get<std::string>("Depth.width").c_str()),
				std::atoi(pt.get<std::string>("Depth.height").c_str()), RS2_FORMAT_Z16,
				std::atoi(pt.get<std::string>("Depth.fps").c_str()));

// Alow recording into file
cfg.enable_record_to_file(files[cnt]);
// Start pipeline with new configuration
pipe_1->start(conf);
```

Is there any option that i can put in these line to turn compression on ? 

```cpp
cfg.enable_record_to_file(files[cnt]);
```

Sorry for the long post.

dorodnic, 2019-07-25:
I understand, this is what I meant by clunky. Unfortunately, there is no overload of `cfg.enable_record_to_file(files[cnt]);` that controls the compression. 
Alternatively, there is always an option to go into the code and hard-code desired value, but it has its own drawbacks.

gachiemchiep, 2019-07-26:
Hello @dorodnic .
Thank you for your information. 
i feel that Hard-coding the code or following your guide may lead to a technique risk.
I decide to purchase a new D435 to replace the D435i. 

Thank you for your time and effort. 
That actually help me a lot.

closed 2019-07-26 by gachiemchiep